olehs/PZEM004T #1: Ability to specify serial port request.

inthebatcave, 2016-03-13:
Thank you for this contribution. I just joined today, because of this. I have almost no idea what I'm doing, and I have more ideas than I have ability.  I'm hoping to piece together a home automation / energy logger and getting my arduino mega to talk to my PZEM004 was my first piece. By pure dumb luck I was able to modify the library file to read from port1, but my end goal is to read from several at once, and I'm not sure where to begin.

olehs, 2016-03-13:
By default, hardware serial port is used only if you specify it's pins (0 and 1). 
If you need to connect multiple devices, you can specify any pair of RX/TX pins except 0 & 1 for them, and the library will use SoftwareSerial to communicate them automatically.

itsjustvenky, 2016-03-14:
I heard we can connect multiple devices on the same Serial pin's and use the IP address to differentiate the device. How true is this ? thanks

inthebatcave, 2016-03-14:
Thank you for your reply, I must not been doing something right then. I had
my device connected to serial1 (18,19), and I wasn't getting any readings.
I guess I need to read up on SoftwareSerial to get a better understanding
of how it works and what it can do.

Sorry for my ignorance, but thank you again!

olehs, 2016-03-14:
> I heard we can connect multiple devices on the same Serial pin's and use the IP address to differentiate the device. How true is this ? thanks
You can connect RX pins of multiple PZEM004s to one RX of your Arduino. But TX outputs of TTL serial devices are driven high when not sending data, so they cannot simply be wired together. You will need a multiplexer to read responses.

inthebatcave, 2016-03-15:
That is inconvenient, seems strange that they are designed to be addressed but they simply can't be put on a common bus. Rather than adding a mutliplexer, perhaps they individual devices can be grounded to a dedicated DO that will only activate when there is an need for a read/write?

olehs, 2016-03-21:
I guess they designed it to support Ethernet on other PZEMxxx devices.
Can you please explain about dedicated DO? Not sure I got the idea.

itsjustvenky, 2016-03-21:
I never saw any PZEM devices with Ethernet port and the energy measurement IC in the device doesn't have any SPI pin's. 

DO : @inthebatcave  says that we need to powerOn one device at a time and read it and switch it off and that using Digital Out pin.

olehs, 2016-09-27:
added ability to use different Hardware Serial ports. See examples

closed 2016-10-12 by olehs