olehs/PZEM004T #4: ESP8266<->PZEM004T

mohsh86, 2016-05-10:
Thanks heaps for your contribution.

I've managed to get it working with ESP8266, software serial works but i get a lot of "stack" info and the frequent Soft WDT reset  interrupts other operations related to wifi connection.

giving hardware serial a shot, believe the RX, TX pins for ESP8266 is 1,3 i've tried those (edited PZEM004T.CPP to accommodate that), i get correct reading sometimes, the rest is just -1.

Is it because am connecting my ESP8266 through USB as well ?

mohsh86, 2016-05-10:
yea that was stupid of me, obviously it was having both USB + TX,RX hooked up

closed 2016-05-31 by mohsh86

tmckc44, 2016-12-29:
Sorry, I want to connect the PZEM004T with ESP8266 too but,it have the problem. It can't read the voltage or anything please you tell how to connect this 2 module pls.