olehs/PZEM004T #5: Energy reading reset

mohsh86, 2016-06-01:
I know there isn't any software reset through serial, but i was wondering how i can handle Energy metering if the meter overflows after "99999" (if that was the max number).

the library returns energy in Wh not kWh, my question, what is the max reading it is going to hit before resetting back to 0, so i can do the math ((99999 - previous reading) + new reading) on every beginning on a month in case the new reading was less than the previous reading

olehs, 2016-06-01:
Sorry, I don't know what is the max reading of this device.
But if it was 99999, the correct math would be (100000 - previous_reading + current_reading)

mohsh86, 2016-06-02:
thanks for the tip, there is something i don't understand, i just contacted the seller and he said the max reading is 9999kWh , i was wondering how is it i am getting Wh not KWh from your library ? it goes back on how many digits your poll is returning ?

olehs, 2016-06-02:
9999kWh needs 5 bytes. The library returns only 3 bytes of data as Energy. 
Accroding to examples, device transmits 3 bytes of Energy and then 2 zero-bytes. But the problem is that Energy is transmitted little-endian, so we can't use those 2 bytes to extend reading.
This means that library's max Energy is aprox.16.8kWh

I guess we need to check the actual behaviour of the meter when it passes 16,78kWh to know data format for sure.

olehs, 2016-06-02:
I had to delete my previous comment because it was completly wrong.
By protocol (see readme) the maximum Energy value is aprox 16777kWh (3 bytes).
So the value 9999kWh seems to be correct.

mohsh86, 2016-06-03:
will keep the meter running and let you know when it turns over, i've been running it for a couple of days now it is 35097.00 Wh, (the decimal point is added by openhab), it is being historized i should be able to plot and see the max it reaches

mohsh86, 2016-06-10:
I had the reading going past 5 digits, it is now reading 156712 Wh

if 9999 kWh is the max reading according to AliExpress, i think it will go 9999999 Wh before it turns over ?!

mohsh86, 2016-08-12:
for who ever is still interested, the number just flipped to the 7th digit, its now > 1000000 and counting

closed 2017-04-28 by olehs

WongKamFu, 2017-04-30:
@mohsh86 Are you still using it? How's the reading? Have you reached the max energy reading value?

mohsh86, 2017-04-30:
I've got a snapshot of today's reading [here ](https://www.dropbox.com/s/6buvn97fqoodpdm/pzem-004t.png)

its 3426456 Wh

WongKamFu, 2017-04-30:
Long way to go I guess :) 
Thanks anyway.

crosofg, 2017-05-05:
Can the energy reading be reset to zero programmatically?

mohsh86, 2017-05-05:
i asked the vendor on aliexpress: nope

crosofg, 2017-05-05:
Any comments on accuracy/reliability of this meter to use in production environment?

mohsh86, 2017-05-05:
if you're planning production environment, better use SDM series from eastron, this one is even hassle to install and has no casing, and no plan for casing

kamiliasana, 2017-08-28:
there is a way for resetting the device, you can put a relay contact in parallel over the switch on the board .
Just program a closed contact for 5 seconds , then release and then again for 1 second and the counter is 0