olehs/PZEM004T #6: Can't pull any readings from Board when USB cable is disconnected!

rushlimelite, 2016-07-28:
Hey everyone, need some help here.
This meter works great (Thanks olehs for the library!), however I can't get any readings when I pull the USB cable from the computer. I have the tx/rx pins assigned as 10/11 and like I said, things work great... until the USB cable is pulled. Do I need to reconfigure the library somehow, or modify some sketch code?

Suggestions anyone?
Thanks,

RushLimeLite

olehs, 2016-07-28:
Hi. Do you still supply +5V to VDD pin after pulling the USB?

rushlimelite, 2016-07-28:
Yes, the board has constant power (+5vdc) at all times. So I don't understand what affect the USB cable between the Arduino Mega and the PC should have on it? Any thoughts?

olehs, 2016-07-29:
Here is a simple test sketch I used to check your issue. It turns on Mega's LED when voltage reading is over 100 Volts. It works both with and without USB connected on my side.

`#include <SoftwareSerial.h>
#include <PZEM004T.h>

PZEM004T pzem(10,11);  // RX,TX
IPAddress ip(192,168,1,1);

void setup() {
  Serial.begin(115200);
  pzem.setAddress(ip);
  pinMode(13, OUTPUT);
  digitalWrite(13, LOW);
}

void loop() {
  float v = pzem.voltage(ip);
  digitalWrite(13, LOW);
  Serial.println(v);
  if (v > 100.0)
    digitalWrite(13, HIGH);
}`

rushlimelite, 2016-07-30:
olehs,
Thank you for your diligence in helping me fix this issue. Upon seeing your sample sketch, I went back and reviewed the code in my sketch. Since they were almost identical, I started looking at the physical connections. While I could get no readings when the USB was unplugged, I quickly realized why...
My power source for the PZEM was not tied to the Arduino. Once I powered the board through the Arduino, everything corrected on it's own. Now, irregardless of if the USB is connected or not, the board is reporting back.
I do have one last "glitch" if you could help me fix. The meter data to the voltage float reads 1.9 volts low (example: Pzem reads 112.4, when my Fluke reads 114.3 volts). That in turn make the amps reading off as well as the wattage consumed. Can you show me how to add that 1.9 volts back into the equation before it goes to serial print? I think fixing this one equation might fix all three. 

Your thoughts?
Thanks again for your time in reply and your help!
RushLimeLite

   

#include PZEM004T pzem(10,11); // RX,TX

olehs, 2016-07-30:
RMS Voltage and Current are caluclated internally by PZEM's processor (see the theory [here](https://openenergymonitor.org/emon/buildingblocks/ac-power-arduino-maths)). And so is the Power.
I don't think you are able to correct anything except Voltage readings in the output.

closed 2016-07-30 by olehs

rushlimelite, 2016-07-30:
ok, so how do I grab those float values from the sketch, modify them before they are sent to the serial print?
The sketch commands are displaying a float value for each right? volts, amps, and watts. So is there a way (for example) to take the float v * 1.019 = pzem.voltage(ip)  or if I misunderstood the code pzem.voltage(ip)* 1.019 = float v?
there has to be a way to allow for corrections?!Am I just wishing here? I don't know yet how all this works, I am just sort of fumbling around trying to understand. Here is the code section for the 3 values in question...
void loop() {
  float v = pzem.voltage(ip);
  if (v < 0.0) v = 0.0;
  Serial.print(v);Serial.print("V; ");

  float i = pzem.current(ip);
  if(i >= 0.0){ Serial.print(i);Serial.print("A; "); }
  
  float p = pzem.power(ip);
  if(p >= 0.0){ Serial.print(p);Serial.print("W; "); }
  
  float e = pzem.energy(ip);
  if(e >= 0.0){ Serial.print(e);Serial.print("Wh; "); }

  Serial.println();
Thanks for your patience with me.Ben

olehs, 2016-07-30:
Try somethin like this
 `float v = pzem.voltage(ip) + 1.9;`

rushlimelite, 2016-07-30:
Thank You!
That did it! I have been so mired in trying to sort things out, I overlooked the simplest method.Must be time for a break.
Thanks again for your help.Regards,Ben