olehs/PZEM004T #10: Multiple Meters

lblacksmith, 2017-03-25:
Greetings! We are currently working on a project involving 2 meters PZEM0004T interfaced with an Arduino Mini Pro. Your library has been of a great help. Thank you!

We set up the system to work with different pins (2 pins each module, al connected to 4 pins of the arduino), and each time we want to make readings from one of the Meters we declare in that local scope a "new" meter, get the readings of voltage, current, power and energy (which read Ok) and then declare the other meter and do the same. We do it like this because it doesn't work as a global instance of the library object. These "recreations" goes around and around inside the loop. 

Although its working, the question is: When jumping from the last reading and declaring a new meter, there is a considerable pause (a little bit larger than the 1 sec timeout). 

We tried to make a ReadPZEM function to fill an array variable so we can send this data over the i2C line. But iterating over the function blinds the second reading. See code attatched (Note: Iteration is not implemented on this code). 

[PZEMTest.txt](https://github.com/olehs/PZEM004T/files/869999/PZEMTest.txt)

I see there is a "Destroy" feature now in your libraries, but I'm not sure if not destroying the module before recreating it is causing the trouble. It should be able to instantiate in different areas of memory. But it doesn't.  Any suggestionns ?

olehs, 2017-03-25:
`//HERE: The switch to the new sensor delays the next reading over a second.
  PZEM004T pzem2(9, 8); // RX,TX module 2
  v = pzem1.voltage(ip); /// <<< Should be pzem2 
  if (v < 0.0) v = 0.0;
  Serial.print(v); Serial.print("V; ");
`
PZEM004T can return as fast as only 2 metrics in 1 second. In the code snippet above you're checking pzem1's voltage again. At this time pzem1 is busy after previous reading.

lblacksmith, 2017-03-25:
Right! Sorry about the mistake. But the same issue is still there, when targeting the meter #2, also back to meter #1 when the loop ends and restarts. I'm going to try and duplicate the library as a separate class and see what happens. I'll let you know. 

Thanks!

olehs, 2017-03-25:
Ok. But I still don't understand what was the problem with global instances of two objects. These classes use no global or static variables, so there shouldn't be a problem with independant multiple instances.

lblacksmith, 2017-04-30:
Greetings, it's me again. I borrowed a logic analyzer from a friend and restarted looking into this. Making another instance does send the request (both of them), and the meters DO reply, but the library can't read the values of the first instance of the library (Meter 1) so it reaches the time out, but it does read the second meter (Meter 2), so it prevails only the latest instanciated object. Making a separate class library, has also the same effect, which is very weird. 

Arduino Serial Monitor Response:
[image]

Analyzer monitoring:
[image]

Code:
[PZEM_2Instances.txt](https://github.com/olehs/PZEM004T/files/966825/PZEM_2Instances.txt)

How ever, if I instantiate within the loop it behaves like it should:

Arduino Serial Monitor Response:
[image]

Analyzer monitoring:
[image]

Code:
[PZEM_Local2Instances.txt](https://github.com/olehs/PZEM004T/files/966830/PZEM_Local2Instances.txt)

I know what's happening but I don't know why is it happening. I'm not really sure, but I believe it might be something to do with the Softwareserial library, but this is just an idea, it might be nonsense. 

I think I'll try to create several SoftwareSerial objects within the Pzem Library, since I know we are going to handle up to 4 Meters. I need the instance to be accessible from other parts of my code (like it should!).

PS: I'm using a single TX pin to communicate with several power meters. It works fine due to the independent address of each meter. 4 Meters would only use 5 pins instead of 8, which is good if you have an Arduino UNO or other low pin count device.  I'm using two microcontrollers to behave as each meter  (since the new meters are on their way), which is why you see the random data. This has been tested with actual power meters but one of the meters is busy (it reads power from a small apartment) and the other had a little accident and got damaged, but the behavior of the library is the same with the microcontrollers as it was with the actual power meters.

olehs, 2017-04-30:
Thanks for all the work you've done. 
I just commited some patch for softserial-based PZEMs. Could you please update and check it? 
Thanks.

lblacksmith, 2017-04-30:
Yes! That fixed it. Wow, so simple. Well I learned something new today!

PZEM004T library is awesome. Thank you for your hard work. It works perfectly!

olehs, 2017-04-30:
Great to here that. Good luck with your project!

closed 2017-04-30 by olehs

ibra85fox, 2018-07-20:
dear @olehs i has pzem004t and give a nagative 1 voltage only .. i has tried everything to get a valid measurement but i can not .. please help me bro

olehs, 2018-07-20:
@ibra85fox 
Please do not duplicate your question in every issue.
I already tryed to answer you question in #13