olehs/PZEM004T #13: Pzem always gives -1 A

crosofg, 2017-04-29:
I have connected Pzem to ARduino Nano through Software SErial. IT alwasy gives -1A as current and all other values are also negative. What could be the reason?

olehs, 2017-05-03:
check connection

closed 2017-05-23 by olehs

ovisopa, 2017-08-02:
Can anyone help me, I have the same issue, all values are -1, and I don;t understand what I did wrong , also I don;t understand what is the IP for, and what IP address is that. I have used one that is free. 

Please check the attached photo to see if anything might be wrong

The settings in the code are 
`PZEM004T pzem(4,5);  // RX,TX (D2, D1) on NodeMCU
IPAddress ip(192,168,1,48);`

The serial monitor reports :
`-1.00V; -1.00A; -1.00W; -1.00Wh; 200
-1.00V; -1.00A; -1.00W; -1.00Wh; 200
-1.00V; -1.00A; -1.00W; -1.00Wh; 200
-1.00V; -1.00A; -1.00W; -1.00Wh; 200
`

!

[whatsapp image 2017-08-02 at 17 19 42](https://user-images.githubusercontent.com/5344791/28879107-eb6f3e70-77a9-11e7-9769-aa400e3f60b4.jpg)

pieman64, 2017-08-02:
@ovisopa in your photo you don't have any electricity going through the coil.

ovisopa, 2017-08-02:
I removed the board from the place where I first installed, there I had connected both 220v wires on the empty connector, and also one of the 220v wires was passing through the coil.

But I get -1 in serial even if I dissconnect the RX / TX wires from ESP8266.. there must be something wrong with my setup, but I don't know what.

I tried to change the IP to multiple values but the same. I also tried to swap RX with TX, every reading is the same -1

pieman64, 2017-08-02:
@ovisopa live wire through the coil presumably?

By convention the IP address of 192,168,1,1 is used but it shouldn't matter.

Do remember the ESP is a 3.3V device so technically you can't feed the Pzem tx pin directly to the rx pin on the ESP. It needs to go through a voltage divider but some people run without one. I wouldn't take the risk for the cost of a couple of resistors.

Are you mixing up GPIO references with D pin references i.e 4, 5 needs to go to D2 and D1?

And finally there is a special ESP version of SoftwareSerial called espsoftwareserial-3.2.3, are you using it?

ovisopa, 2017-08-02:
When I first tried to flash the ESP I had some errors regarding the software serial library, but I fixed it after downloading another library, now I only get a warning when flashing the ESP.

I think the problems is regarding 5v on the PZEM004 vs 3.3v on the ESP, If I change R17 as in the photo https://github.com/pcgayan/PZEM004T/blob/5d39a972395d6405ac0dd86805f8d2fbb30f036b/examples/SampleWiringDiagram.jpg

..will it work without using a voltage divider ? or a TTL level converter, I think I purchased in the past some modules, to convert levels from 3.3 to 5v .. I need to check my inventory :D

`WARNING: library PZEM004T claims to run on [avr] architecture(s) and may be incompatible with your current board which runs on [esp8266] architecture(s).`

ovisopa, 2017-08-05:
To eliminate even one possible problem, I used an Arduino Pro Micro (Leonardo) that is runing at 5v, but I still can't get any reading, anything I can do to test if the module I purchased is working ?

Here's a video with the new setup 

https://www.youtube.com/watch?v=zVeowB13q4g

Thank you.

pieman64, 2017-08-05:
@ovisopa you seem very reluctant to show the coil. Do you have live, rather than neutral, through the coil?

ovisopa, 2017-08-05:
I'm not trying to hide anything, the coil was hanging down, no wire through it, but it should have display the voltage, and 0w for the power.

Even with the ESP, I think it should display 0v even if no wire is attached to the AC connector, right ? It should be able to work when there is no power from the mains, but to be sure that is not the case TODAY I connected the AC also, but nothing through the coil

[image]

olehs, 2017-08-05:
Yes, it should show 0V when there is no current. ButNo, it's IC is powered by AC voltage, so without 220V it returns an error (-1)

ovisopa, 2017-08-05:
I have also tried to use an ESP USB programmer to connect to the PZEM module and use the chinese software provided by [PEACEFAIR](https://www.aliexpress.com/store/product/PEACEFAIR-AC-100A-Electric-power-monitoring-and-communication-module-power-meter-power-energy-Volt-Ammeter-with/1773456_32405328185.html?spm=2114.12010608.0.0.2222514ax62h7C) but there are only COM1 - COM4 options available, and my USB serial converter is shown on COM5, the other ports are already used :(

Should have worked with this adapter, is there a way to test using an USB to serial adapter like this ? 
[image]

pieman64, 2017-08-05:
@ovisopa it has just been explained to you by @olehs why you are seeing 0V. Wire up the coil but don't just randomly take "one of the 220v wires".

olehs, 2017-08-05:
@pieman64 actually I ment coil is not needed) just 220V.
@ovisopa can you please show RX/TX connections in your Arduino setup?

ovisopa, 2017-08-05:
@olehs , in my last test, using the 5v arduino, I had 220v connected, but I didn't know the IC is powered by AC, so when there is no AC (mains power down) I will have to make some modifications to my database, to not store -1 but 0v ... if I will make it work after all :)

Guys, any ideea what I can do to test the module if it's working are welcome. i posted the video so it will be easier for you to see the connections, if I should tell any more detail, please ask, I will post back.

I think I will purchase another one - or two modules (anyway I wanted to measure all 3 phases, not just one, but for the beggining I wanted to test one module, to see what result I get ..but not that lucky yet :( )

Thank you.

ovisopa, 2017-08-05:
@olehs I will connect again the module, and make a picture, but here is a screenshot from the video posted above . Yellow wire was on RX, white wire was on TX of the PZEM, but I tried to swap them also, in case I haven't connected them correctly

[image]

olehs, 2017-08-05:
RX of Arduino (according to your code - it's pin 2) should be connected to TX of PZEM and vise versa.
AC power has to be connected to PZEM and must be on.

ovisopa, 2017-08-05:
here is another video showing the code, the connections, and the coil (with no wire through it), I used a simple voltage checker not the multimeter, as I was holding the phone with one hand, I can't touch the AC connectors with the other hand to show you there was 220vAC present

https://www.youtube.com/watch?v=9GOqFOwBYno

olehs, 2017-08-05:
Well, wiring seems correct. 
Dont' know what can be wrong here. You should try PEACEFAIR's software with your USB/TTL adapter. Just change it's COM number in Windows Device Manager settings to one of COM1 -COM4

ovisopa, 2017-08-05:
the other coms show as already used, I can only asign COM2, but COM2 is disabled in PEACEFAIR software.

And now I updated the code, to print on serial the response it gets from the module, because the 0v we seen  before with this new code, for arduino leonardo, was made by if (v < 0.0) v = 0.0.. so there was -1 all the time, not 0v, exactly like the response I got on the ESP

[image]

Any other ideeas are welcome. 

Thank you anyway :) If I can't make it work, I will post back in a few weeks, when I will receive the new PZEM modules, I will buy 2 more these days.

olehs, 2017-08-05:
One more idea, 
Try pins 8 and 9 on Arduino for RX/TX (don't forget to set them in sketch)

pieman64, 2017-08-05:
@olehs yes you are right that without current going through the coil then it will still show the voltage, as screenshot below:
[image]

olehs, 2017-08-05:
@ovisopa If you know your COM ports 1-4 are not used by any present hardware, you can change your USB adapter's port number to any of them, even if DevManager states they are in use

ovisopa, 2017-08-06:
I have tried the example HardwareSerial and moved the serial wires to RX and TX pins on the Arduino Pro Micro, and with this small change in the code, I have some valid readings!!!! Finally. Now I need to see why it didn't work on the other pins. I will try now with the Wemos and the same code, to see if I can make it work.

1. In the picture bellow you can see how I connected the Arduino with the PZEM module, pin 2 of arduino to pin 3 of PZEM, pin 3 of arduino to pin 2 of PZEM. 5v from VCC of arduino to pin 1 of PZEM. AC voltage is connected on the other connector, but through the coil I don;t have any wire, it should display the voltage and 0w for power

2. The sketch I used in this Arduino Pro Micro was copied from this forum, I just modified it to actually show the actual value received from PZEM instead of 0v, so it shows -1 for each option, exactly like the other code shown when I tested with the ESP8266.

[code]#include <SoftwareSerial.h> // Arduino IDE <1.6.6
#include <PZEM004T.h>

PZEM004T pzem(2,3);  // RX,TX
IPAddress ip(192,168,1,1);

void setup() {
  Serial.begin(9600);
  pzem.setAddress(ip);
}

void loop() {
  float v = pzem.voltage(ip);
  //if (v < 0.0) v = 0.0;
  Serial.print(v);Serial.print("V; ");

  float i = pzem.current(ip);
  if(i >= 0.0){ Serial.print(i);Serial.print("A; "); }

Serial.print(i);Serial.print("A; ");
  
  float p = pzem.power(ip);
  if(p >= 0.0){ Serial.print(p);Serial.print("W; "); }
  
Serial.print(p);Serial.print("W; ");  
 
  float e = pzem.energy(ip);
  if(e >= 0.0){ Serial.print(e);Serial.print("Wh; "); }
  
Serial.print(e);Serial.print("Wh; ");

  Serial.println();

//  delay(1000);
}[/code]

3. The library is the one downloaded from https://github.com/olehs/PZEM004T

UPDATE. I just tried now the code from the example, which uses the hardware serial and connected the pins to RX and TX pin of arduino, and now I can see the voltage !!! Really really happy I can see something now :) But I need to make it work with software serial too, as on the ESP8266 (Wemos D1 mini) that is what I need, or it should work using this line on ESP too ? 

[code]
PZEM004T pzem(&Serial1);
[/code]

Readings:
`229.20V; 0.00A; 0.00W; 4062.00Wh; 
229.00V; 0.00A; 0.00W; 4062.00Wh; 
229.10V; 0.00A; 0.00W; 4062.00Wh; 
228.90V; 0.00A; 0.00W; 4062.00Wh; 
229.00V; 0.00A; 0.00W; 4062.00Wh; 
228.90V; 0.00A; 0.00W; 4062.00Wh; 
229.00V; 0.00A; 0.00W; 4062.00Wh; 
`

PS. Is there a way to reset the total energy with a command ? I want o reset it each month. Also pushing the CLR button does nothing, I tried 4 sec long press , double press, single normal press .. but the total it still 4062Wh

[image]

Thank you.

pieman64, 2017-08-06:
I asked PeaceFair and they don't have a software reset for Energy. Clear button does work.  Press for X seconds and then press again.  Forget how many seconds X is.

ovisopa, 2017-08-06:
Now, back to the ESP8266 (Wemos D1 mini) I have modified the code to like this, and connected the serial pins to the RX and TX pins of the Wemos D1, it should work  ?

```
#include <SoftwareSerial.h>
#include <PZEM004T.h> // https://github.com/olehs/PZEM004T
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

const char* ssid = "Cisco08753";
const char* password = "Extr3m3WD";

//PZEM004T pzem(4,5);  // RX,TX (D2, D1) on NodeMCU
PZEM004T pzem(&Serial1);
IPAddress ip(192,168,1,1);

void setup() {
  Serial.begin(9600);
  pzem.setAddress(ip);
```
[image]

vortigont, 2017-08-06:
@ovisopa This would not work. You can try hardware serial on Wemos D1 by using other pins, not the ones connected to onboard usb-ttl.
Connect wemos's pins D7,D8 to pzem's Tx,Rx pins respectively. !Important! Do not connect pzem's Vcc for a while, first power up Wemos, let it boot up and only after that connect Vcc to +5 v line. Otherwise it won't boot.
A code to use this setup can be something like this:

HardwareSerial hwser(HWSERIAL_PORT);
PZEM004T pzem(&hwser);
void setup() {
hwser.swap(); //swap hw_serial pins to gpio13(rx),15(tx)
// other code....
}

olehs, 2017-08-06:
@ovisopa 
> Not all pins on the Leonardo and Micro support change interrupts, so only the following can be used for RX: 8, 9, 10, 11, 14 (MISO), 15 (SCK), 16 (MOSI).
See https://www.arduino.cc/en/Reference/softwareSerial

Is your level converter a two-way converter? Try ESP without it first. On you photo RX goes to RX. Must be RX - TX.

ovisopa, 2017-08-06:
@pieman64 Thanks for the tip on how to reset the energy, I didn;t tried long press followed by a short press, it worked, after my second try, about 5 - 6 seconds for the long press, and than quick a short press, the energy value was cleared, as you can see in the attached photo (it's the apache access.log on my BananaPi)

[image]

@vortigont thank you for joining in this thread, it took me a while (+30 minutes :) ) to find out what should I use for HWSERIAL_PORT, and after some time I found the source of HardwareSerial.cpp  and there was my answer :) UART0 **And now I finally have a valid reading on the ESP too** :D

But now I have another problem, as you mentioned the ESP will not boot with VCC connected on the PZEM, to me it only boots if I remove the TX/RX wires from D7/D8 when powering up the ESP, if the wires are removed it boots up, than after ~3 seconds I insert them and I get my readings sent to my bananaPi server.

Is there an easy solution to this new issue? I want to place 3x PZEM and 3xESP8266 (if I can't connect more meters to a single ESP, I haven't reached that stage, didn't even searched to see if it's possible for one ESP to communicate with 3 PZEM) but the ideea is to monitor the power consumption for all the 3 phases, the devices will be mounted in my mains fuse panel (old ceramic fuses, not new automatic fuses, there is plenty of space), I will probably power the ESP from an UPS, to be able to monitor even if the AC is down but I want it to work without the need to remove the SERIAL wires on each boot

@olehs Now that I can communicate with the PZEM module, I also have the confirmation that the level converter works, I tried first the HardwareSerial without the Level converter, and it worked, but after that I connected through the level converter, and it still works, so the level converter is ok. **Should I use it ? Does it help, or I should connect the serial wires directly to the PZEM?** If it worked for a few minutes with the wires connected directly, can there be an issue in the long therm if I won't use a level converter ? I'm looking for simplicity, if it's safe without the level converter, I won't use one.

Now I will try again with software serial and the ESP, maybe it will work. I will double check the wires TX to RX and RX to TX.

Thank you very much guys, I'm so glad I finally have some valid readings from this meter.

pieman64, 2017-08-06:
Software serial will work with the WeMos and Pzem becuase that's how I have it set up. I wouldn't run 5V into the ESP for any longer than is necessary.

vortigont, 2017-08-06:
@ovisopa 

> it took me a while (+30 minutes :) ) to find out what should I use for HWSERIAL_PORT
Actually sometimes the simplest answer is the right one - you could have just put 'HWSERIAL_PORT' (yes, just like this), it is predefined to UART0. That's good that you have your reading now :)

> Is there an easy solution to this new issue?
This could be resolved by connecting D8 pin via transistor to the pzem, a simple pnp in common collector (emitter follower) mode would do. By the way this would also allows you to get rid of the level-convertor for TX pin. RX pin doesn't need it in the first place.

As for controlling 3 pzems via one ESP, that is an interesting topic. I do not see any reasons why this cannot be done via 3 swserial's. It's more software task that hardware. Or, maybe, adding a multiplexer board could do the trick with hw-serial. I hadn't tried it. I have two pzem meters, but one is permanently hooked to the house's main, so can't check it.
Working with swserial, especially with 3 at a time, might result in lots of watchdog timeouts with ESP. You might require this [patch](https://github.com/vortigont/PZEM004T/commit/c20dfab4efd2bdd7b35a89cc5cfb1d547b00d13e#diff-a9209d9d24927496e5e83481d4c02ef7) to resolve it.

ovisopa, 2017-08-07:
@vortigont 

> Actually sometimes the simplest answer is the right one - you could have just put 'HWSERIAL_PORT' (yes, just like this), it is predefined to UART0. That's good that you have your reading now :)

It didn't work for me, using this line HardwareSerial hwser(HWSERIAL_PORT); I got an error when I tried to compile, as I remember it was HWSERIAL_PORT was not defined in this scope.

Today I can't do more test as I have alot of work to do, and I already spend many hours today playing with this device :) 

I still have the module on my office desk, no wire through the coil, so far I monitor the voltage only, I will try to move it somewhere else to have a wire trough the coil and monitor the energy also. I added the voltage graph to my dashboard (is still under construction) and I really like the results :)

[image]

vortigont, 2017-08-07:
> It didn't work for me, using this line HardwareSerial hwser(HWSERIAL_PORT); 

Ah-h, my bad, this was just a copy-paste from my code, I forgot that I have my own def's... Sorry for the confusion.
Your panel looks nice, is that a MajorDoMo?

ovisopa, 2017-08-07:
I used to build CMS / websites in the past, for more than 10 years, so building my own dashboard was the best solution for me, as I wanted to include **many different devices** and features in this dashboard, I don't know if I could do that with the other already available home automation dashboards.

It's basically a website, an responsive interface that adapts nicely to phone screen, table and PC, so far I haven't made it available from outside of my local network, but I want to be able to access it from outside using a VPN, I installed the VPN server on the same BananaPi that also hosts this dashboard, and that is the brain of the entire system, every device pushes the stats to this bananaPi. I have some issues with the VPN but I will fix them in time ... 

With this dashboard I already did a few things I really wanted to have, in the same interface, and not a different application for each feature, there are still alot of things that I want to add, so far I did those things:

- [x] Monitoring temperature / humidity
- [x] Monitoring PC's CPU load / RAM / Power on time
- [ ] Monitoring disk usage
- [ ] I have 2 dedicated servers with hundreds of websites hosted, I really wanted to have the Email QUEUE, CPU Load and RAM usage, I almost finished this bash script but it doesn push the informations to my bananaPi yet, I only added to my dashboard some static graphs for this purpose.
- [x] Monitoring my LAN and send notification to my phone (trough PushBullet) when a new UNKNOWN device connects to my LAN/Wifi. 
- [x] The network section is splitted in 3 tabs, Unknown devices, 24/7 devices, and known devices. For the 24/7 devices I also set up the notifications, if one device is set as a 24/7 device and is not online for more than 3 minutes, it sends an notification to my phone
- [x] I can turn off (SLEEP) my PC's from this dashboard, and also **I can turn them on** if the motherboard supports WOL, so far I tested this with success on my workstation as this is the most important, but I needed a way to also turn off some PC's in my workshop, that sometimes I forgot them ON and I didn't want to have a VNC service running on those too, each PC's is controlling a CNC machine, and I wanted to have as fewer softwares installed on them. All this stats reporting and turning off of the PC's is done through a single batch script setup to run each 5 minutes in Windows Task Scheduler, it reports the status of the CPU/RAM ..etc and than it receives an response from the server, that can be, so far, OK or SLEEP, if it gets the SLEEP response it turns off the PC.
- [x] Monitoring temperature and load of other RaspberryPi devices I have in my lan, one is running the firmware of FlightRadar24 (ADSB Receiver) and one is Running LiveATC, I also can listen the LiveATC stream directly from this interface, there is a very small PLAY button in that section if you check the videos bellow
- [X] Turning ON/OFF a 10 meters led strip I have in my office
- [ ] I want to use this dashboard to control the heating inside my house, instead of buying a ready made room thermostat I will use a simple temperature sensor and the part that controls the  gas heater I was told it's quite simple, it just needs to close a circuit to turn it on, and open that circuit to turn it off. I did so far just the slider with the desired temperature, and temperature readings, but did not yet implemented the the way to send the command to turn on/off the gas heater.
- [ ] for my workshop side of the dashboard, I will include an air pressure sensor for the compressor, smoke detectors, dust sensor ..etc
- [ ] other plans are to include a static photo with the last movement grabed by the security cam, clicking the photo should start the camera stream to view it live
- [ ] many other more plans .. but I'm quite off topic now with this long post :) 

Some videos I sent to some friends with the dashboard progress:

https://www.youtube.com/watch?v=NOmANB3GqVc

https://www.youtube.com/watch?v=xPsQverAP6A

https://www.youtube.com/watch?v=05s6N_XOGLc

vortigont, 2017-08-08:
@ovisopa  That's really impressive! Ever considered to make it open-source? :)

ovisopa, 2017-08-11:
Not really, at least not in the form it is right now, every modification needs to be made directly by modifying the code, it's quite a mess right now as I tried many charts libraries, I need to clean up the source a lot.. but maybe, when I will have it done to a stage where **anyone can use it without codding** 

My plan is to make a user based dashboard, after login in to the dashboard each user should be able to control/view only the sensors or machines that he is interested in
For example my wife won't need to see all my hosting servers stats, or sensors in the workshop, but she will want to be able to change the room temperature, maybe get notification about opened roof windows when it rains - I forget from time to time to close the one in the bathroom :) 
My brother is also sharing the same office with me, he doesn't want to see the hosting server stats, but he wants to control the office lights/temperature, and also the sensors in the workshop

And to be back on topic, in the mean time I was looking for a way to **reset remotely** the total energy consumed, I wanted to use an transistor/optocoupler to activate the CLR button on the PZEM module, I got some other advices like:

1. use a small reed relay (I was looking for micro relays, but the price was too high, but reed relays are very very cheap on ebay, I will buy a few just to have them :) )
2. don't reset the energy using hardware, but do it in software ! this was a very smart advice, instead of dealing with the electronics I could setup the script that is saving the information in database, to substract the energy consumed until past month
If I will start logging from 1 september 2017, I will save the value I will get from the module untill 1 september, and starting from 1 octomber the script should substract the total energy consumed until 30 september, than starting from 1 november the script should substract the total energy consumed untill 31 octomber, including september. This could be done easily and automatically in the script that is saving this information in database. 
3. use optocupler or transistor ... I will try all those to see which one works.

vslinuxdotnet, 2018-01-23:
Hello, some question, is possible to reset the energy by serial port command ?
Thank You

pieman64, 2018-01-23:
@vslinuxdotnet there is no command to do that and when we asked Peacefair they said they had no plans to add it. Others have said that you could wire up your own circuit to do it but as might know it's not just a simple press of the button.

vslinuxdotnet, 2018-01-24:
Ok, I will try to do it with optocupler. Thank you

ibra85fox, 2018-07-20:
@ovisopa 
dear i has Arduino mega 2560 and PZEM004T(V2.0) and i try everything but i still get -1 for the voltage ...
tell me please what i do to get a valid data 
thanks bro

olehs, 2018-07-20:
1. Check you TX/RX pins are connected properly.
2. Check you PZEM is powered with 5V.
3. Check you have connected AC source to PZEM and it's switched on.
What pins do you use on MEGA and how do you initialize PZEM object (soft, hard)?

ovisopa, 2018-07-21:
@ibra85fox follow the steps  @olehs wrote above, if still not working I think it's a good ideea to post also your entire code.

PS. AC wires should be connected

gasparformador, 2020-03-27:
Hello!

I had the same issue and I solved:
1- Check your board because there are different versions:
PZEM004T
https://github.com/olehs/PZEM004T
PZEM004T v3-0 (mine)
https://github.com/mandulaj/PZEM-004T-v30

2-Check 5 Volts arrive to the board.

3-REMEMBER! Arduino TX goes to PZEM RX

4-When you plug the AC power, a led turn on close to connector)

ENJOY!

MAROUANE98, 2021-04-27:
Hello everyone , please I just baught pzem-004t V3 and when I interface it with esp8266 it doesn't work.Just RX on pzem blinks and TX is not . Pleaase any help !!!!!