olehs/PZEM004T #14: commit b7c55247e84e77d1f5da757f6094254a59d74d78 breaks the library for ESP's

pieman64, 2017-05-15:
@olehs the commit you made on the 30th of April for PZEM004.cpp of:

line 136    if(_isSoft)
line 137        ((SoftwareSerial *)serial)->listen();

stops compilation of your library on ESP's.

What does the commit do and can we simply revert back to the earlier cpp?

The compiler error for ESP's with the 2 lines included is:

```
C:\Users\Public\NewArduinoSketchesAndLibraries\libraries\PZEM004T-master\PZEM004T.cpp: In member function 'bool PZEM004T::recieve(uint8_t, uint8_t*)':

C:\Users\Public\NewArduinoSketchesAndLibraries\libraries\PZEM004T-master\PZEM004T.cpp:137:37: error: 'class SoftwareSerial' has no member named 'listen'

         ((SoftwareSerial *)serial)->listen();
```

olehs, 2017-05-15:
You could try using [this library](https://github.com/plerup/espsoftwareserial) or just revert to previous version.
This commit allows to use multiple PZEMs at once with one Arduino. Default Arduino's SoftwareSerial supports multiple instances and I don't know why ESP's one doesn't,

olehs, 2017-05-15:
BTW, looks like library I specified in my previous comment is just the default library of ESP for Arduino.
Maybe you just need to update your ESP8266 libraries?

pieman64, 2017-05-15:
@olehs I will check through your comments. Do you not use the PZEM with an ESP?

olehs, 2017-05-15:
No, I use it with Arduino Mega and Ethernet shield. I actually prefer wired solutions )

pieman64, 2017-05-15:
I thought as much becuase I don't think I have any outdated core libraries.
Could you perhaps modify the commit just to cover Arduino board types?

vortigont, 2017-05-22:
@pieman64 I do not see this error. I was able to build with the last commit and EspSoftwareSerial under platformio and under Arduino IDE with espressif SDK 2.0
@olehs It would be great to add library.json to your project, so it could be easily tracked.

olehs, 2017-05-23:
@vortigont Did you install EspSoftwareSerial manually or it was installed as a part of ESP SDK (using board manager)?

vortigont, 2017-05-23:
@olehs I used version from [git](https://github.com/plerup/espsoftwareserial)
Just checked, indeed arduino bundled version for ESP8266 platform throws compile error.

olehs, 2017-05-23:
The author of ESP8266 SDK agreed to update his library
https://github.com/esp8266/Arduino/issues/3239

outrullers, 2017-05-27:
Thank you!
I have got the same compile error with ESP8266, and this helped me a lot!

P.S. @olehs I deleted my misplaced issue

pieman64, 2017-06-29:
@olehs I'm using the master branch of the Arduino core for ESP and I can confirm all is now fine.

closed 2017-07-04 by olehs