olehs/PZEM004T #15: kw alarm

goku75, 2017-05-21:
I can not set kw alarm. Can you have an example of how to do this?

olehs, 2017-05-23:
Hi. There is setPowerAlarm method. You can use it like this
`pzem.setPowerAlarm(ip, 500);`

olehs, 2017-05-23:
btw, how are you going to use this alarm?

goku75, 2017-05-23:
To communicate i used arduino mega. But now the only thing I care about is that it is when I overtake the 3 kw of absorption.

olehs, 2017-05-23:
According to docs, PZEM's screen flashes when power is over threshold. But PZEM004T doesn't have a screen. This is why I wonder).
But you can check power in your Arduino sketch by yourself.

vortigont, 2017-05-23:
There is a version with screen and it blinks indeed. There is also a place and pads for buzzer on the PCB inside, so maybe it can even make some noise )

goku75, 2017-05-23:
I have the model with led screen and buzzer. I need the mega arduin program to set power-arm to 3 kw. Already I now read arduino values.

goku75, 2017-05-23:
I use this to read the data:
#include <SoftwareSerial.h> // Arduino IDE <1.6.6
#include <PZEM004T.h>

PZEM004T pzem(10,11);  // RX,TX
IPAddress ip(192,168,1,1);

void setup() {
  Serial.begin(9600);
   Serial1.begin(9600);
  pzem.setAddress(ip);
}

void loop() {
  float v = pzem.voltage(ip);
  if (v < 0.0) v = 0.0;
  Serial.print(v);Serial1.print("V; ");

  float i = pzem.current(ip);
  if(i >= 0.0){ Serial1.print(i);Serial1.print("A; "); }
  
  float p = pzem.power(ip);
  if(p >= 0.0){ Serial1.print(p);Serial1.print("W; "); }
  
  float e = pzem.energy(ip);
  if(e >= 0.0){ Serial1.print(e);Serial1.print("Wh; "); }

  Serial1.println();

//  delay(1000);
}

goku75, 2017-05-23:
How do I enter this?
pzem.setPowerAlarm(ip, 20);

for 3kw

olehs, 2017-05-23:
add this at the end of setup()
`pzem.setPowerAlarm(ip, 3);`

goku75, 2017-05-23:
Ok it works, the display blinks exceeded the 3 kw but does not sound :(
I'll check.
thank you

closed 2017-05-24 by olehs