olehs/PZEM004T #19: Feature: Add power factor calculation

vortigont, 2017-05-27:
Pls, consider adding a feature to calculate Power-Factor out of current UIP meterings.
I've made some tests for different loads and seems it works pretty fine most of the time.
[Here](https://github.com/vortigont/espem/blob/master/examples/webstat/sampleschart02.png) is a sample graph.
I've [implemented](https://github.com/vortigont/PZEM004T/commits/master) this a new method to the lib and an example code. If it looks reasonable, I can make a PR.
Thanks.

olehs, 2017-05-27:
Thanks. I'll take a look at it soon.
I see you've added _yield_ in reading method. Does it solve issue described here https://github.com/olehs/PZEM004T/issues/8 ? Are there any side effects?

outrullers, 2017-05-27:
Hello!
I use this library with ESP8266 (Wemos D1 mini)

errors after library update:
`C:\...\Arduino\libraries\PZEM004T-master\PZEM004T.cpp: **In member function 'bool PZEM004T::recieve(uint8_t, uint8_t*)':**

C:\...\Arduino\libraries\PZEM004T-master\PZEM004T.cpp:137:37: error: **'class SoftwareSerial' has no member named 'listen'**

         ((SoftwareSerial *)serial)->listen();`

olehs, 2017-05-27:
@outrullers see https://github.com/olehs/PZEM004T/issues/14

vortigont, 2017-05-28:
Yes, yield() makes a great difference for esp platform. Putting it between consecutive calls to pzem outside of the library solves the watchdog problem, but esp is still unresponsive to networking during serial IO.
Putting it inside the receive() lib call solves this issue too.
As for the side effects - I do not see any. Now I'm running it in software-serial mode, do not see any read errors, no data skips, so far so good.

vortigont, 2017-07-01:
@pieman64 To your [comment](https://github.com/vortigont/PZEM004T/commit/c20dfab4efd2bdd7b35a89cc5cfb1d547b00d13e#commitcomment-22861581), it is not so simple unfortunately. While it is easy to spot the problem in such a cases like only one pure active load - it's either switched off or on and you have reading current value 0 or >0. But it becomes more complicated if power meter monitors main line of your house with lots of loads simultaneously. You always have non zero current&power values and pf is usually <1 because of lots of switching PSU's. PF calculation works fine for steady loads but could give spikes in both directions if measurement had taken place during heavy transient power changes.  And which of the pzem readings you trust more - the current or the power? That is an open question to discuss and it would be nice to have more feedback on this.
For my meter project I'm [fixing](https://github.com/vortigont/espem/blob/3038e0fb9d6e25a342f9b3b2b4f950d315bf3788/espem/functions.ino#L113-L114) current value based on power readings if pf spikes to >1, but for the library implementation I think that it's better not to tackle actual pmeter readings but let user decide how deal with it.
I'm open to suggestions on this topic.

pieman64, 2017-07-01:
@vortigont yes you guessed right we are not currently using the Pzem for "whole house" monitoring. In fact some of the monitoring is down at just 70mA and 1 or 2W. Not sure what the accuracy is at this level when the Pzem only offers integer Watts etc.

In our sketch as well as v & I > 0 we have p > 30. Anything less than 30W gives a widely inaccurate pf.

vortigont, 2017-07-02:
@pieman64  To rise sensitivity for such a low power solutions I've tested setup with increased number of wire turns over CT and correcting read values in sketch, it could help but requires some calibration.

closed 2021-09-22 by vortigont