olehs/PZEM004T #22: Question: Using the PZEM004T with an ESP8266

pieman64, 2017-07-07:
@vortigont and others that are using the PZEM with an ESP.

1. Are you using  a voltage divider between ESP RX and PZEM TX rather than messing around with the change to R17 on the PZEM?

2. Are you using hard or soft serial for the interface and which pins are you using on the ESP?

vortigont, 2017-07-10:
@pieman64 
1. actually one of my meters runs without any divider/level_converter, just +5v connected to pzem Vcc.
It was a temporary solution at first, just for a test setup, but than I forgot about it :) And it runs like this about 6 months. I think that voltage dropout on the optocoupler's LED is enough to protect  ESP pin. BUT  it's definitely not a recommended setup! For another meter I've set a simple level converter with FET's that I bought somewhere on AliExpress.

2. I'm running both options. For an old meter I'm still using sw-serial and it runs fine with [this patch](https://github.com/vortigont/PZEM004T/commit/c20dfab4efd2bdd7b35a89cc5cfb1d547b00d13e#diff-a9209d9d24927496e5e83481d4c02ef7R150).
Another meters runs hw-serial using swapped UART to pins 13,15. Default pins doesn't work because of the usb-ttl converter. The problem with swapped pins is that ESP doesn't boots up after power loss if attached directly to pzem 'cause of the pin 15 pull-up high, I use bipolar transistor to workaround this.

pieman64, 2017-07-12:
Excellent post @vortigont

pieman64, 2017-07-16:
@vortigont do you occasionally get "bad" data and even ESP resets from the Pzem?
By adding ESP.wdtFeed() and some other hardware hacks we think we have the resets covered. 

Our Pzem updates a Smartphone app and yesterday morning  one of the graphs showed some strange values (see below). Voltage dropped from 240V to 120V for 40 minutes!

[image]
It's normal for our voltage to drop from 240V to 230V during the day which I believe is the electric company trying to cope with high demand for aircon etc but it's the first time I have seen anything like 120V.
Our Pzem takes readings at 2s intervals but the graphs and database only keep average values over a 60s period. The 1 minute average value fell gradually rather than a sudden drop, so 240, 234, 217, 209, 193, 187, 175, 169, 164, 150, 128, 120, 120 [40 minutes at 120V] 177, 185, 201, 217, 240, 240.

Maybe it's not the Pzem and the electric company had some failure, what do you think based on your experience?

vortigont, 2017-07-19:
@pieman64, I have almost no problems with esp resets. yield() added to the PZEM lib does the trick.
As for the PZEM data read errors I had some indeed. One of these was that sometimes (although very seldom) I saw that "Energy" counter increased for about a hundred watts for no reason while all other readings (current, power) showed no spikes at the same time. I have no idea what was that all about.
And the other case - a couple of times PZEM readings returned ALL zero values. That was not an read error (for that lib returns -1.0f) but true zero values for all params U,I,P,W. Although that situation is logically possible, but for my project saving this results to the Database poisoned stat query and resulted in a wrong averaged stats. So I had to implement a [workaround](https://github.com/vortigont/espem/commit/b8d194ad8a8b9e3a22bcf694ff7cb65b8209746c) for that exact case.
As for you case - that really looks strange, never faced anything like this. Are you sure that those gradually decreasing values are actual reading but not some averaging done with an app widget? If you monitor current and power values at the same time than you can try to compare and check if that was an actual voltage drop or a read errors maybe? But that kind of voltage drop is definitely out of any possible fluctuations for any energy company. I doubt it.

pieman64, 2017-07-19:
@vortigont yes as per my post the plots are 1 minute average values taken from 30 off readings at 2 second intervals. So with the readings of 120V it's possible the Pzem was alternating between 240V and 0V throughout each 1 minute period. However that doesn't really explain the gradually decreasing and increasing plots before and after the "120V" period. Well it could be explained if it was only dropping to 0V for a short period of the 1 minute average but strange that it get progressively more "0V" and progressively less "0V" at the end of the "120V" period.

The app also plots WATTS and Energy costs with the WATTS data also being visible above the VOLTAGE in the screenshot I posted. The WATTS and Cost do not show any variation from the assumption of 240V throughout the period so suggests a reading error.

So you have had all zero values for U,I,P,W. Brief power failure or bad readings?

vortigont, 2017-07-20:
@pieman64 Do you have the logs what was the actual data that time, not the averaged graph?
And another question do you have any checking in you code for the value returned by pzem lib?
I mean in case of read error lib returns -1.0f, if you try to average -1 with 240 volts it may actually give you resulting ~120 volts. I remember receiving lot's of -1.0f errors when doing frequent polling.
And why using such a low interval for polling (2 secs) if anyway all you got is a 1 minute average?

As for all zero values it was not power failure. But even if it was "Energy" counter can't return, say '10500' than '0' and than increase to '10501' again. So i think it was some error in the pzem controller itself.

pieman64, 2017-07-24:
@vortigont no I don't have the 2s logs, just the 1 minute "average" logs.
Yes my code only takes values >= 0.
2s is overkill but I actually calculate my own energy (and cost) so if a device is powered up it will only be 2s before energy and cost is corrected.
The authors of the Android part of the app (I just do the C++ for the MCU) believe their averaging shouldn't give the values shown in the graph if the data included 0V.

pieman64, 2017-08-20:
@vortigont or @olehs do you happen to know the maximum current **input** that the Pzem's can accept? We appear to be overloading ours when the ESP is running from a 2A supply.

DimChik2903, 2017-10-05:
@pieman64 Hello! Could you show your program for the pzem with Blynk? I do not know how to make them work. Thank you in advance!

pieman64, 2017-10-05:
@DimChik2903 it's a little off topic but yesterday I published a Blynk project for the ACS712. If you check out my ESPecoMon © repository it should give you an idea how to incorporate the Pzem into a Blynk app.

DimChik2903, 2017-10-09:
@pieman64 I can not figure out how to write a sketch to work with the hardware serial (RX/TX pin) on ESP8266. Could you add an example? Thank you in advance!

pieman64, 2017-12-18:
@DimChik2903  hopefully @olehs will not mind us providing details of our Discourse Community site which covers our development of the PeaceFair product along with other energy monitoring systems. If you  are still struggling with hardware serial please feel free to post a question at https://esppromon.com/t/peacefair-pzem-004t-energy-meters/24

KPeare, 2018-04-15:
@vortigont I've tried all sorts to upload to my NodeMCU 12E but nothing seems to work, I keep getting an error. I have pasted below.

Can anybody help????

E:\DOCUME~1\Cube\LOCALS~1\Temp\arduino_build_955589\sketch\PZEMDisplay.ino.cpp.o:(.text.setup+0x10): undefined reference to `PZEM004T::setAddress(IPAddress const&)'

E:\DOCUME~1\Cube\LOCALS~1\Temp\arduino_build_955589\sketch\PZEMDisplay.ino.cpp.o: In function `HardwareSerial::begin(unsigned long)':

E:\Documents and Settings\Cube\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/HardwareSerial.h:75: undefined reference to `PZEM004T::setAddress(IPAddress const&)'

E:\DOCUME~1\Cube\LOCALS~1\Temp\arduino_build_955589\sketch\PZEMDisplay.ino.cpp.o:(.text.loop+0x10): undefined reference to `PZEM004T::voltage(IPAddress const&)'

E:\DOCUME~1\Cube\LOCALS~1\Temp\arduino_build_955589\sketch\PZEMDisplay.ino.cpp.o:(.text.loop+0x1c): undefined reference to `PZEM004T::current(IPAddress const&)'

E:\DOCUME~1\Cube\LOCALS~1\Temp\arduino_build_955589\sketch\PZEMDisplay.ino.cpp.o:(.text.loop+0x24): undefined reference to `PZEM004T::power(IPAddress const&)'

E:\DOCUME~1\Cube\LOCALS~1\Temp\arduino_build_955589\sketch\PZEMDisplay.ino.cpp.o:(.text.loop+0x28): undefined reference to `PZEM004T::energy(IPAddress const&)'

E:\DOCUME~1\Cube\LOCALS~1\Temp\arduino_build_955589\sketch\PZEMDisplay.ino.cpp.o:(.text.loop+0x43): undefined reference to `PZEM004T::voltage(IPAddress const&)'

E:\DOCUME~1\Cube\LOCALS~1\Temp\arduino_build_955589\sketch\PZEMDisplay.ino.cpp.o: In function `loop':

E:\Documents and Settings\Cube\My Documents\Arduino\PZEM004T-master\examples\PZEMDisplay/PZEMDisplay.ino:13: undefined reference to `PZEM004T::current(IPAddress const&)'

E:\Documents and Settings\Cube\My Documents\Arduino\PZEM004T-master\examples\PZEMDisplay/PZEMDisplay.ino:15: undefined reference to `PZEM004T::power(IPAddress const&)'

E:\Documents and Settings\Cube\My Documents\Arduino\PZEM004T-master\examples\PZEMDisplay/PZEMDisplay.ino:18: undefined reference to `PZEM004T::energy(IPAddress const&)'

E:\Documents and Settings\Cube\My Documents\Arduino\PZEM004T-master\examples\PZEMDisplay/PZEMDisplay.ino:21: undefined reference to `PZEM004T::PZEM004T(unsigned char, unsigned char)'

E:\DOCUME~1\Cube\LOCALS~1\Temp\arduino_build_955589\sketch\PZEMDisplay.ino.cpp.o: In function `_GLOBAL__sub_I_pzem':

E:\Documents and Settings\Cube\My Documents\Arduino\PZEM004T-master\examples\PZEMDisplay/PZEMDisplay.ino:21: undefined reference to `PZEM004T::PZEM004T(unsigned char, unsigned char)'

E:\DOCUME~1\Cube\LOCALS~1\Temp\arduino_build_955589\sketch\PZEMDisplay.ino.cpp.o: In function `__static_initialization_and_destruction_0':

E:\Documents and Settings\Cube\My Documents\Arduino\PZEM004T-master\examples\PZEMDisplay/PZEMDisplay.ino:24: undefined reference to `PZEM004T::~PZEM004T()'

E:\DOCUME~1\Cube\LOCALS~1\Temp\arduino_build_955589\sketch\PZEMDisplay.ino.cpp.o: In function `_GLOBAL__sub_D_pzem':

E:\Documents and Settings\Cube\My Documents\Arduino\PZEM004T-master\examples\PZEMDisplay/PZEMDisplay.ino:24: undefined reference to `PZEM004T::~PZEM004T()'

collect2.exe: error: ld returned 1 exit status

Using library SoftwareSerial at version 1.0 in folder: E:\Documents and Settings\Cube\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\SoftwareSerial 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

vortigont, 2018-04-18:
@KPeare Looks like an issue in your sketch, it would be more informative you could provide your sketch. But, pls, do not paste it here, use for e.x. https://paste.ee/ or something like this.

KPeare, 2018-04-22:
Hi, and thank you for your email @vortigontUnfortunately I have not had much spare time for this project.I have moved on a little with it, that being that I have added a few files into the project library but the sketch still will not complete. The sketch I am using is the PZEM004_1 taken from the github olehs site.
I guess I don't have the right library files at the moment for the sketch to complete. 
Thanks for your support. 
Regards 
Kevin

Vadzz-K, 2018-05-04:
Hi all!
I have a long time working ESP8266 (ESP12-E) with PZEM-004t. I using hardware UART (and OTA for firmware update). To work with hardware UART, the configuration was used as indicated in the example:
`HardwareSerial hwser(UART0);
PZEM004T pzem(&hwser); // Connect to PZEM via HW_serial`
I periodically updated Arduino IDE, libraries and boards. And recently I needed to make changes to the firmware of ESP8266 (changes not related to the PZEM-004t module). All successfully compiled, updated the firmware on the OTA, and the ESP8266 stopped running - more precisely, it constantly rebooted the watchdog timer:
`ets Jan  8 2013,rst cause:8, boot mode:(3,6)
 
wdt reset
load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld`
If you exclude the PEME004T.h library and associated functions from the project, then the ESP starts and runs normally. I spent a lot of time figuring out this problem, eventually it all worked after I determined the use of hardware UART in the sketch as follows:
`PZEM004T pzem(&Serial);
IPAddress ip(192,168,1,1);`
Thank you. Maybe someone will benefit from this information.

vortigont, 2018-05-11:
@Vadzz-K  actually it's the same setup. "Serial" object you are using must be created first.
Could you, pls, specify esp core version and SDK you've faced this issues? Was it a default Arduino's IDE update or a trunk/git version os ESP8266 core? Looks like something has changed in recent core's updates to the default objects. Maybe we need to fix the example but it must be done in a way to keep compatibility with an old IDE/core versions.
Thanks for the report!

Vadzz-K, 2018-05-11:
Hi, @vortigont !
New issue created - [Using ESP8266 Hardware UART for PZEM-004t](https://github.com/olehs/PZEM004T/issues/40)

closed 2018-09-15 by olehs