olehs/PZEM004T #23: Can Energy be set to Zero in software?

pieman64, 2017-07-24:
I know the Energy can be set to zero with the key in the LCD display or pressing the reset button on the board but I wondered if there was also a software hack?

If the Pzem is used for remote monitoring there will be nobody around to reset the Energy when each electric bill arrives and is validated as OK.

I have sent a message to Peacefair but just wondered of anyone had found a hack?

pieman64, 2017-07-25:
Peacefair have replied and said there is no software hack to reset Energy, shame.

vortigont, 2017-07-26:
I think this could be done with a help of one more optocoupler connected to the button and any mcu pin controlled via software.

olehs, 2017-08-05:
@pieman64 I guess we can close the issue?

pieman64, 2017-08-05:
Yes @olehs ok to close.

closed 2017-08-05 by olehs