olehs/PZEM004T #30: power output from code not Equal to power from formula

xang555, 2017-11-15:
thanks for you create this library.  but i have some problem with power value that i get from example code
here my output:

[image]

i Calculated power  by using P = U * I.  so i test with my current and voltage data that i get but Calculated  value from formula not Equal or adjacent  power from example code output. 

power output Equal to power from formula?.

i'm use nodemcu broad.

olehs, 2017-11-15:
This formula is true only for active load.
See https://en.wikipedia.org/wiki/AC_power#Active.2C_reactive.2C_and_apparent_power

pieman64, 2017-11-15:
@xang555 many "whole house" energy meters are not particularly accurate at very low power usage so that might be some of your problem. The other issue is that **real power** is not simply U * I.

If the load has a power factor of 1 (or 100%) then it would be U * I but some loads have a pF of 0.7 etc.

xang555, 2017-11-15:
thanks for your fast reply. i'm understand and close this issue AC not DC.  i'm stupid hahahah

closed 2017-11-15 by xang555