olehs/PZEM004T #33: Found PZEM004T's Warning when compile on ESP8266/ESP32

TridentTD, 2018-02-18:
When i compile on ESP8266/ESP32 , it occurs warning like the folowing message.

----------------------------------------------------
WARNING: library PZEM004T claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (esp8266) architecture(s).
----------------------------------------------------

Maybe because of "architectures" option in file "library.properties"

---------------------------------------------------------
name=PZEM004T
version=1.0
author=Oleg Sokolov
maintainer=
sentence=Enables communication Peacefair PZEM-004T Energy monitor
paragraph=
category=Sensors
url=https://github.com/olehs/PZEM004T
architectures=avr      <------------------  change to *
---------------------------------------------------------

Thank you.

olehs, 2018-02-19:
There was a problem with SoftwareSerial library version in ESP (see #14), but it seems to be fixed now.
I guess I can add this platform now.

vitalyxxl, 2018-03-30:
Добрый день , Олег ! 
Наконец-то удалось собрать этот скетч . Постоянно ругалось на SoftWareSerial 
У меня плата HELTEC ESP32 .  Что-то не пойму как подключить PZEM004 к этой плате . 
+5V и GND подключил .   На моей плате есть  выводы   TX и RX  . Но как их указать в скетче ? 
Я только начинаю изучать Ардуино :)

olehs, 2018-03-30:
https://github.com/olehs/PZEM004T/tree/master/examples/PZEM_ESP8266HwSerial
Там в скетче указано куда что подключать

vitalyxxl, 2018-03-30:
Большое спасибо !  А можно с Вами как-то консультироваться в этих делах ? Я хочу сделать управляемую розетку с мониторингом .... и связь держать по LoRa . Для этого имеется и роутер DRAGINO LG-01 . Может быть Вы уже копали в эту сторону ?  Может через Скайп или Viber ? 
+375 29 101-888-0 Виталий (Минск.Беларусь)

olehs, 2018-03-30:
Сомневаюсь, что буду полезен, т.к. я изначально писал библиотеку только для Ардуино, а с ESP ее подружили другие ребята с гитхаба. Даже тот пример не я делал.

vitalyxxl, 2018-03-30:
Понял . Но Вы хоть кто-то из русскоязычных :)    кто понимает в этом .

olehs, 2018-03-30:
Могу предложить зарегаться на https://majordomo.smartliving.ru/forum/index.php
Там есть и железячники, и программисты. Я там тоже активно стараюсь помогать.

vitalyxxl, 2018-03-30:
Большое спасибо !

closed 2018-05-19 by olehs