olehs/PZEM004T #34: Suggestion

merlinthemagic, 2018-02-20:
I was banging my head against the wall trying to get this lib to work for over a day on an ESP8266 . It turned out the reason was simply that the PZEM004T must be connected to AC before readings are possible.

I wanted to write a step by step wiki to help others, but it seems the wiki is disabled (odd behavior as it redirects to the code tab). If you enable Wikis i will write up a short summery

olehs, 2018-02-20:
It would be great! I've created a home page. Please, try it now

merlinthemagic, 2018-02-20:
Hi,

Either i need to be added as a collaborator or you can open up the wikis for public editing: https://help.github.com/articles/changing-access-permissions-for-wikis/

MM

olehs, 2018-02-20:
Done

merlinthemagic, 2018-02-20:
https://github.com/olehs/PZEM004T/wiki/Basic-Setup

Please edit as needed.

closed 2018-02-20 by merlinthemagic

pieman64, 2018-02-20:
@merlinthemagic you might want to mention a voltage divider on the return from the Pzem's TX to the MCU's RX when using a 5V feed for 3.3V MCU's. Espressif have suggested that their MCU's are "5V tolerant" but for the cost of a couple of resistors it's probably not worth the risk.

merlinthemagic, 2018-02-20:
Hi,

@pieman64 good suggestion. feel free to add / edit as you please it’s a public wiki. :)