olehs/PZEM004T #36: esp8266 2.4.1 problem

xsiriusx, 2018-03-23:
using the library esp8266 2.4.1 the pzem004t probe does not read the watt and watt / h values. Restoring the library to 2.3.0 works great. Problem of library esp8266 or this library? sorry for my english

xsiriusx, 2018-04-06:
just for information the problem is that randomly some readings are -1, as if communication was missing through the serial

olehs, 2018-04-06:
Sorry, I can't check this setup right now (don't have this hardware). Maybe someone who has ESP and PZEM can check it.
But since the problem is random, I have to ask if you're testing it on default sample sketch? Did you change timeout setting in it?

vortigont, 2018-04-06:
esp8266 2.4 core introduced a lot of changes that affects timing-dependent code. SoftwareSerial lib has some issues too (plerup/espsoftwareserial#63). You can try building with most recent SoftwareSerial lib code from git or use esp8266 with hardware-serial. I'm running it without any issues with a 2.4 core and SDK:2.2.1 in hwserial mode.

closed 2021-06-02 by olehs