olehs/PZEM004T #37: Added ESP32 & hardware serials support

mikekgr, 2018-04-17:
Dear Sir,
If you agree, please expand your nice library in order to support the ESP32 microcontroller both as ESP32 family ( at platforms )  and the hardware serials that they have.

Thanks and Best Regards,
Mike Kranidis

olehs, 2018-04-17:
Hello, Mike.
I just order an ESP32 board on Aliexpress. I'll see what needs to be added to the library as soon as I receive it.

olehs, 2018-05-19:
please check the latest updates (see esp32 example)

mikekgr, 2018-05-19:
Dear @olehs ,
thanks a lot. I will test it in the coming days and I will let you know.
Many thanks.
Best Regards,
Mike Kranidis

banatus, 2018-07-06:
Dear @olehs 
Pzem004T works very well with EPS32 such Doit or (16,17 pin ) Serial 2 . I use wemos lolin32 without 16,17  pin (http://www.instructables.com/id/ESP32-With-Integrated-OLED-WEMOSLolin-Getting-Star/) .
 I have some trouble with Serial 2 . So I red http://support.iteadstudio.com/support/discussions/topics/11000014882 about How to change or add the pins : My code 
HardwareSerial Serial1(1);
#define SERIAL1_RXPIN 25 or 12 
#define SERIAL1_TXPIN 26 or 13 

//HardwareSerial Serial2(2);     // Use hwserial UART2 at pins IO-16 (RX2) and IO-17 (TX2)
//PZEM004T pzem(&Serial2);
PZEM004T pzem(&Serial1);
IPAddress ip(192,168,1,1);
It did work at all , I got many notification errors : 
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a5c
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400815dd  PS      : 0x00050034  A0      : 0x400d3552  A1      : 0x3ffc0600  
A2      : 0x00000000  A3      : 0x00000009  A4      : 0x00000016  A5      : 0x3ffc05d0  
A6      : 0x00000000  A7      : 0x00000003  A8      : 0x00000001  A9      : 0x400864dc  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x0000cdcd  A13     : 0x00000001  
A14     : 0x00060021  A15     : 0x00060c23  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x400d3552  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400815dd
So I changed to the normal

HardwareSerial Serial2(2);     // Use hwserial UART2 at pins IO-16 (RX2) and IO-17 (TX2)
PZEM004T pzem(&Serial2);
IPAddress ip(192,168,1,1);
It seems look good but the status is 
Connecting to PZEM...
Connecting to PZEM...
Please give me some advice.
How to map the pins for Wemoslolin32 
Thank you very much
Banatus S

olehs, 2018-07-06:
To use Serial1 (likein you example) you have to reinit it with specified pins in your setup().
Try adding this code to setup()
`Serial1.end();`
`Serial1.begin(9600, SERIAL_8N1, SERIAL1_RXPIN, SERIAL1_TXPIN);`

banatus, 2018-07-08:
Hi olehs 
I tried to use serial1 and I couldn't make it in the first start.
So I decided to use the simple way to use serial2 and changed pin in  HardwareSerial.cpp file instead.
void HardwareSerial::begin(unsigned long baud, uint32_t config, int8_t rxPin, int8_t txPin, bool invert)
{
     if(_uart_nr == 0 && rxPin < 0 && txPin < 0) {
        rxPin = 3;
        txPin = 1;
    }
    if(_uart_nr == 1 && rxPin < 0 && txPin < 0) {
        rxPin = 9;
        txPin = 10;
    }
    if(_uart_nr == 2 && rxPin < 0 && txPin < 0) {
        rxPin = 16;
        txPin = 5;
    }
    _uart = uartBegin(_uart_nr, baud, config, rxPin, txPin, 256, invert);
}
from rxPin=16 txPin=17 to  rx=16 and tx=5 of Wemoslolin32 http://www.instructables.com/id/ESP32-With-Integrated-OLED-WEMOSLolin-Getting-Star
It was sucessful. I got  the information correctly.
227.30V; 0.00A; 0.00W; 12.00Wh; 
Thank you 
Banatus S

closed 2018-08-17 by olehs