olehs/PZEM004T #38: ESP8266 Can't read anything

CrackerNo1, 2018-04-17:
I use PZEM004T connect to Arduino it work smoothly, but when I connect to ESP8266 it isn't working. It only display **0.00V;** on Serial Monitor Arduino IDE.
VCC-3.3V
RX-D8(GPIO15)
TX-D7(GPIO13)
GND-GND
I try to swap pin TX-RX
This is my code.
`#include`` <SoftwareSerial.h> // Arduino IDE <1.6.6
#include <PZEM004T.h>
//#include <ESP8266WiFi.h>
PZEM004T pzem(13,15);  // (RX,TX) connect to TX,RX of PZEM
IPAddress ip(192,168,1,1);

void setup() {
  Serial.begin(115200);
  pzem.setAddress(ip);
}

void loop() {
  float v = pzem.voltage(ip);
  if (v < 0.0) v = 0.0;
  Serial.print(v);Serial.print("V; ");

  float i = pzem.current(ip);
  if(i >= 0.0){ Serial.print(i);Serial.print("A; "); }
  
  float p = pzem.power(ip);
  if(p >= 0.0){ Serial.print(p);Serial.print("W; "); }
  
  float e = pzem.energy(ip);
  if(e >= 0.0){ Serial.print(e);Serial.print("Wh; "); }

  Serial.println();

//  delay(1000);
}`

vortigont, 2018-04-22:
You need to connect pzem Vcc to +5 volts, or use level converter, or replace smd resitor for pzem rx pin to a value about 430 Ohm.

hasanbasri1993, 2018-05-01:
how @vortigont has solved with change resistor ?

vortigont, 2018-05-13:
@hasanbasri1993, pls see Wiki page
https://github.com/olehs/PZEM004T/wiki/Make-PZEM-3.3v-compatible

CrackerNo1, 2018-07-10:
If I use soft serial and I replace R17-470 Ohm, I connect RX-GPIO4, TX-GPIO13, VCC-3.3V. I think there is nothing wrong here but it can't be working, at Serial of Arduino IDE always display 0.00V;
Please help me!!

SeByDocKy, 2019-05-11:
Hi,
Maybe you will be able to help me ...
I have (had) one PZEM-004T coupled with a nodeMCU V3 where VCC=+5V of the nodeMCU. It worked fine.... Then I decided to add a DHT22 in +3.3V...... Catastroph !!!!! If I got the temperature read correctly... now no more current & power reading.... everyting to 0. Even when I removed the DHT22... still no reading except the voltage ..... I am afraid I killed my PZEM-004T.... but I don't know why..... ANy idea ?

mudassar791, 2019-07-10:
Can any body explain why to modify pzem 004t with 1kohm resistor to its coupler?
While nodemcu esp8266 based can provide 5V. Can we use nodemcu with pzem004t with out this modification?

CrackerNo1, 2019-07-10:
> Can anybody explain why to modify pzem 004t with 1kohm resistor to its coupler?
> While nodemcu esp8266 based can provide 5V. Can we use nodemcu with pzem004t with out this modification?

We need 5V power supply for PZEM, in the communication, there are 2 optocouplers, NodeMCU-3.3v, if we keep 1k resistor it can't work, so we need to modify resistor to lower resistor value.
The last important thing is you need to use <b> Arduino_ IDE <1.6.6 </b>
[image]

mudassar791, 2019-07-10:
> > Can anybody explain why to modify pzem 004t with 1kohm resistor to its coupler?
> > While nodemcu esp8266 based can provide 5V. Can we use nodemcu with pzem004t with out this modification?
> 
> We need 5V power supply for PZEM, in the communication, there are 2 optocouplers, NodeMCU-3.3v, if we keep 1k resistor it can't work, so we need to modify resistor to lower resistor value.
> The last important thing is you need to use Arduino_ IDE <1.6.6 
> [image]

But VIn nodemcu is also 5v, why we cant use them instead of this alteration?

CrackerNo1, 2019-07-10:
> But VIn nodemcu is also 5v, why we cant use them instead of this alteration?

Have you ever tried to keep 1k resistor? Was it works?

SeByDocKy, 2019-07-10:
I am with the 1K resistor mod for my PZEM-004T V2. Notice the PZEM-004T don't need this mod since the new chip can work @3V3. The 1K mod is required for the V2 because if not the RX/TX will be running @5V. The ESP8266 officially don't support the 5V but only 3V3.

mudassar791, 2019-07-10:
Can we say? Pzem-004t is nit compatible with esp8266 in most cases, becuase my device pzem-004t V3 is absolutely ok but cant connect to nodemcu 8266, any guide or instructions?

vortigont, 2019-07-10:
@mudassar791 you can run pzem and esp8266 with 5V and it will work just fine in this setup. Optocoupler's LED Vf drops voltage to a nearly safe ~4V, although it is all at your own risk!
I had this setup running at 5 volts for more than a year without any issues.

closed 2021-06-02 by olehs