olehs/PZEM004T #40: Using ESP8266 Hardware UART for PZEM-004t

Vadzz-K, 2018-05-11:
Hi all!
I have a long time working ESP8266 (ESP12-E) with PZEM-004t. I using hardware UART (and OTA for firmware update). To work with hardware UART, the configuration was used as indicated in the example:
`HardwareSerial hwser(UART0);`
`PZEM004T pzem(&hwser); // Connect to PZEM via HW_serial`
I periodically updated Arduino IDE, libraries and boards. And recently I needed to make changes to the firmware of ESP8266 (changes not related to the PZEM-004t module). All successfully compiled, updated the firmware on the OTA, and the ESP8266 stopped running - more precisely, it constantly rebooted by the watchdog timer, like this:
`ets Jan  8 2013,rst cause:8, boot mode:(3,6)`
`wdt reset`
`load 0x4010f000, len 1264, room 16`
`tail 0`
`chksum 0x42`
`csum 0x42`
`~ld`
If you exclude the PEME004T.h library and associated functions from the project, then the ESP starts and runs normally. I spent a lot of time figuring out this problem, eventually it all worked after I determined the use of hardware UART in the sketch as follows:
`PZEM004T pzem(&Serial);`
`IPAddress ip(192,168,1,1);`
Thank you. Maybe someone will benefit from this information.

>@vortigont:
 @Vadzz-K actually it's the same setup. "Serial" object you are using must be created first.
Could you, pls, specify esp core version and SDK you've faced this issues? Was it a default Arduino's IDE update or a trunk/git version os ESP8266 core? Looks like something has changed in recent core's updates to the default objects. Maybe we need to fix the example but it must be done in a way to keep compatibility with an old IDE/core versions.
Thanks for the report!
P.S. pls, create a new issue for this case. It would be easier to track it.

I can not say at exactly the exact moment when the design specified in the example stopped working because my device has been running for about six months and during this period I periodically updated Arduino IDE, libraries and boards (including esp8266 core) - but there was no need to compile the firmware for my device. I saw this bug just now: Arduino IDE 1.8.0, Esp8266 by Esp8266 Community version 2.4.1....

KPeare, 2018-05-12:
Hi, and thank you for your reply. I am very new to using Arduino so please excuse me.
the unit I have is a wemos.cc LoLin new NodeMcu V3.
I have downloaded a default example 'Blink' and connected a LED just to check it out.
I have then copied the files for PZEM004t from github. Then tried download but that's when I get the error, "Error compiling for board NodeMCU 1.0 (ESP-12E Module).".

At this stage I don't really know where I'm going wrong.

Regards
Kevin

vortigont, 2018-05-13:
@Vadzz-K 
I've just tried to reproduce the issue and found that:
 - crashes is not reproduced with an example sketch, it builds and runs fine at leat for several hours, I also tried to rebuild and run my own project based on PZEM004T lib. It also runs fine for now.
 - it doesn't builds at all with the code you provided (PZEM004T pzem(&Serial);) without defining a "Serial" class first. Build error is
`
/mnt/str/dev/MyProj/testhwser/src/PZEM_ESP8266HwSerial.ino:19:21: error: no matching function for call to 'PZEM004T::PZEM004T(HardwareSerial&)'
PZEM004T pzem(Serial);           // Attach PZEM to hwserial
`
I've tried both core 2.4.1 and an old 2.2.0

But I've used https://platformio.org/ IDE for building and switching different core versions. So the issue might be in way arduino IDE build runs or maybe there is something special in your sketch? Could you, please, try to build and test just an example sketch from PZEM004T lib and check if it crashes too? Also would be nice to try and build it with your class option to check if builds at all?
Do you use any debug output via Serial()?
If your code is publicaly available I can check it with Platformio build.
I'll try to reproduce it all under arduino IDE if time permits.

vortigont, 2018-05-13:
@KPeare 
you must check your logs to see the actual build error. pls, search for basic operations with Arduino IDE

Vadzz-K, 2018-05-13:
@vortigont 
Well, I'll try to check the specified points if time permits. In addition, to test I need to disassemble my device, which works fine in the system now.
In my sketch I do not use debugging on the serial port. But I using Blynk, 1-Wire temp. sensors and the following libraries are used: BlynkSimpleEsp8266.h, ESP8266WiFi.h, WiFiClient.h, EEPROM.h, OneWire.h, DallasTemperature.h, ESP8266WebServer.h, ESP8266mDNS.h, PZEM004T.h.
My sketch is not a secret, but at the moment I would not want to publish it ... Sorry ...

vortigont, 2018-05-14:
Found a PR esp8266/Arduino#2807 that now allows disabling global serial obj's through define's. So I I've tested both lib's example sketch and using global obj serial (PZEM004T pzem(&Serial);), both seems fine to me. Pls, let me know if you could find more details about your issue or exluding other libs tests.

KPeare, 2018-05-14:
Hi @vortigont
Firstly, thank you for all the work that you are putting into this issue.
Secondly, unfortunately I don't have a lot of spare time to put your findings into action. As soon as I can I will try and get things up and running. Regards kpeare@aol.com

SumeriLal, 2019-08-28:
Use this library : https://github.com/SumeriLal/Arduino/tree/master/MasterPZEM

closed 2021-06-02 by olehs