olehs/PZEM004T #44: Support for PZEM-0016

Stompie, 2018-06-28:
Good day,

I absolutely love your PZEM-004T Library and I am in desperate need for help on a library for the PZEM-016 (New model) which uses RS485 as a communication protocol. I don't know where to search or who to turn to anymore. I am willing to pay you if you could help modifying your library a little bit to support the PZEM-016 with arduino. The link for the new model and manual with the codes is here: https://www.aliexpress.com/item/PZEM-016-AC110-220V-100A-Power-Monitoring-Voltage-Current-Active-Power-Meter-With-RS485-Interface-Communication/32876406004.html

Please let me know soon if you will do this or assist me in making this library. Thank you very much

olehs, 2018-07-05:
Hello, @Stompie 
What board are you going to use as RS485 master ?

rowanbradley, 2018-11-03:
Did you succeed in getting an ESP8266 to work with the PZEM-016? I am thinking of going down the same road, to get me a low cost power meter which can transmit readings to my PC via WiFi. I also need to read the generation from my solar PV array. If you have any code, design or other advice that would help me, it would be great if you could share it.

Thanks - Rowan

Stompie, 2018-11-03:
Hi Rowan,

I could not remember if I tested it with an ESP8266. I did get it to work on arduino's and am sure with very little tinkering it should work for a esp. https://pastebin.com/VE4MbrjY. Hope it helps. :-)

morganflint, 2020-07-26:
According to the first post, PZEM-016 has the same protocol as PZEM004T **V3.0**, described here:
https://innovatorsguru.com/wp-content/uploads/2019/06/PZEM-004T-V3.0-Datasheet-User-Manual.pdf

This library is for the previous version, with a different protocol, but in this other issue:
https://github.com/olehs/PZEM004T/issues/40
SumeriLal pointed to another library, specific for 004T V3.0, which should be compatible also with 016:
https://github.com/SumeriLal/Arduino/tree/master/MasterPZEM

morganflint, 2020-07-26:
Another library for PZEM004T-V3.0:
https://github.com/mandulaj/PZEM-004T-v30