olehs/PZEM004T #45: Current Direction

brunohorta82, 2018-07-06:
Hello, thanks for this great lib :), my question is if this lib is already implement the current detection Hack on REVP pin.

Thanks

olehs, 2018-07-06:
Sorry, I don't know about this hack. Do you have some links to explain what does that hack do?

brunohorta82, 2018-07-06:
you can see more information about this on https://github.com/apreb/eNode

This hack is useful when you have solar painel to produce energy and if you produce more than you consume this energy goes to the provider energy network, if you can detect de current direction you can add the minus or plus sign when you measure de power.

olehs, 2018-07-06:
Ok. Now I understand.
So, depending on the pin's state, Current and Power should change there signs.
But what about Energy, does PZEM subtract it's value when direction is reversed?

brunohorta82, 2018-07-07:
I think not, at this moment I try detect de signal using the photocopolator, I created a Pzem Wi-Fi shild to use this feature.
https://github.com/brunohorta82/ESP8266_PZEM_EMONCMS_MQTT

brunohorta82, 2018-09-15:
Hello, please close this issue, to read the current signal it's only need read the GPIO state after make the hack :).

Thanks

closed 2018-09-15 by brunohorta82