olehs/PZEM004T #47: How to use HardwareSerial, example PZEM_ESP8266HwSerial.ino not working

mikekgr, 2018-08-15:
Dear Sirs,
Hardware: ESP-12F/ESP8266
To get the printings of the values ( voltage, current, energy etc ) I use OLED display.

Softwarte: Arduino 1.8.4 running on Windows 7 64bit PC,  having ESP8266 Arduino core 2.4.2 installed.

I need to have PZEM004 connected to ESP8266 hardware serial means ESP8266 TX pin (GPIO01) to PZEM004 RX pin and ESP8266 RX pin (GPIO03) to PZEM TX pin.
Well, IF I use software serial ( instead hardware serial ) to ESP8266l means ESP8266 TX pin (GPIO13) to PZEM004 RX pin and ESP8266 RX pin (GPIO12) to PZEM TX pin all are ok
If I use the HardwareSerial nothing is "printing" at my OLED.
In both cases the rest of sketch are the same.
Please tell me what to do or what is possible wrong. It will be a big help if you just test the example PZEM_ESP8266HwSerial.ino and see if it working on your end.

Thanks and Best Regards,
Mike Kranidis

olehs, 2018-08-15:
Hello, @mikekgr.
Have you noticed the line with _swap()_ in PZEM_ESP8266HwSerial.ino ?
https://github.com/olehs/PZEM004T/blob/807b65fa631114319eb5d0745c6ac4705c9a847f/examples/PZEM_ESP8266HwSerial/PZEM_ESP8266HwSerial.ino#L23

Try to remove (comment) it and check again.

mikekgr, 2018-08-15:
Dear @olehs,
thanks a lot for your reply. In my test sketch I have the following lines regarding the PZEM operation:
`
#include <PZEM004T.h> //https://github.com/olehs/PZEM004T
HardwareSerial hwserial(UART0);     // Use hwserial UART0 at pins GPIO1 (TX) and GPIO3 (RX)
PZEM004T pzem(&hwserial);           // Attach PZEM to hwserial
IPAddress ip(192,168,1,1);

float voltage_b = 0.0;
float current_b = 0.0;
float power_b = 0.0;
float energy_b =0.0;

void setup()
{
pzemrdy = pzem.setAddress(ip);
}

void loop()
{
    float v = pzem.voltage(ip);
    if(v >= 0.0){
      voltage_b = v; //V
    }
}
`
Is there something wrong or something missing in my simplified sketch above? What do you think?
Thanks a lot.

mikekgr, 2018-08-16:
@olehs 
Dear Sir,
I have tried to debug my problem regarding the not working hardware serial, using software serial.
In the sketch that is follows, I measuring the total duration that the function "ReadAndPost" takes. It is very strange that in almost each iteration of this function, the "function duration time" is increasing steadily without any excuse that I can think of except something wrong either my PZEM004 of software serial library. Can you test the follow sketch and let me know if at your end has the same result? 
Thanks in advance!

My sketch:
```
/// PLEASE PUT YOUR DATA AT THE LINES: 44, 61, 67, 68, 
///#define hardwareserial
#define softwareserial

#define COMP_DATE __DATE__

//#define BLYNK_PRINT Serial
#define HARDWARE_LED 2 /// WeMos D1 mini Hardware Led at GPIO02

/// V1 Labeled Value widget, Voltage           V1 ///
/// V2 Labeled Value widget, Current           V2 ///
/// V3 Labeled Value widget, Power             V3 ///
/// V4 Labeled Value widget, Energy            V4 ///
/// V5 Labeled Value widget, current Date/Time V5 ///
/// V14 Terminal widget                       V14 ///

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <BlynkSimpleEsp8266.h>
#include <PZEM004T.h> //https://github.com/olehs/PZEM004T
///PZEM004T pzem(&Serial);                                        /// use Serial
IPAddress ip(192,168,1,1);

#ifdef softwareserial /// using softwareserial ///
#include <SoftwareSerial.h>
PZEM004T pzem(12,13);  // (RX,TX) connect to TX,RX of PZEM  PZEM004T pzem(12, 13);
#endif

#ifdef hardwareserial ///using hardwareserial ///
HardwareSerial hwserial(UART0);     // Use hwserial UART0 at pins GPIO1 (TX) and GPIO3 (RX)
PZEM004T pzem(&hwserial);           // Attach PZEM to hwserial
#endif

/// NTP dependencies and declarations///
#include <TimeLib.h>
/// it is used above /// #include <ESP8266WiFi.h>
/// it is used above /// #include <WiFiUdp.h>

// NTP Servers:
static const char ntpServerName[] = "gr.pool.ntp.org";

int timeZone=3; /// this is initial time zone.
/// That is the blynk way to take TZ ( not the case in this sketch ) This takes feed from time input widget V21, V22, V23, V24
/// t.getTZ_Offset() numeric value in seconds that holds the selected at the widget time zone offset e.x. Time zone offset: 10800
WiFiUDP Udp;
unsigned int localPort = 8888;  // local port to listen for UDP packets

float voltage_blynk=0;
float current_blynk=0;
float power_blynk=0;
float energy_blynk=0;

unsigned long functionDuration = 0;
/// character arrays that keep formated current Date and Time ///
char currentDateTime[25];

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char blynk_auth[] = "PUT YOUR AUTH HERE";

[... 247 lines cut ...]
// send an NTP request to the time server at the given address
void sendNTPpacket(IPAddress &address)
{
  // set all bytes in the buffer to 0
  memset(packetBuffer, 0, NTP_PACKET_SIZE);
  // Initialize values needed to form NTP request
  // (see URL above for details on the packets)
  packetBuffer[0] = 0b11100011;   // LI, Version, Mode
  packetBuffer[1] = 0;     // Stratum, or type of clock
  packetBuffer[2] = 6;     // Polling Interval
  packetBuffer[3] = 0xEC;  // Peer Clock Precision
  // 8 bytes of zero for Root Delay & Root Dispersion
  packetBuffer[12] = 49;
  packetBuffer[13] = 0x4E;
  packetBuffer[14] = 49;
  packetBuffer[15] = 52;
  // all NTP fields have been given values, now
  // you can send a packet requesting a timestamp:
  Udp.beginPacket(address, 123); //NTP requests are to port 123
  Udp.write(packetBuffer, NTP_PACKET_SIZE);
  Udp.endPacket();
}
/*-------- END of NTP code ----------*/

```

olehs, 2018-08-16:
This is how PZEM works. It can return only 2 readings (any values) within ~1 second and then becomes busy.
See the examples: there are no delays in code, but measurements are printed with delay to Serial.
That is why PZEM004T library has default timeout of 1000ms. You can decrease it, but you will receieve error reading (-1) while PZEM is busy.

olehs, 2018-08-16:
To make it clear, you should change this function to measure only PZEM timings without blynk
```
void ReadAndPost()
{
unsigned long temp;
temp = millis();

float v = pzem.voltage(ip);
if(v >= 0.0){ voltage_blynk = v;} //V

float i = pzem.current(ip);
if(i >= 0.0){ current_blynk = i;}  //A

float p = pzem.power(ip);
if(p >= 0.0){ power_blynk = p;} //kW

float e = pzem.energy(ip);
if(e >= 0.0){  energy_blynk = e;} ///kWh

functionDuration = millis() - temp; // moved this line here

Blynk.virtualWrite(V1, voltage_blynk);
Blynk.virtualWrite(V2, current_blynk);
Blynk.virtualWrite(V3, power_blynk);
Blynk.virtualWrite(V4, (energy_blynk/1000));
}
```

mikekgr, 2018-08-16:
@olehs ok I will do right now and I will report it the result back here. Thanks

mikekgr, 2018-08-16:
@olehs still the same, see the attached "screenshot" 
[image]

olehs, 2018-08-16:
Well, it's hard to say what causes this extra delay in multitasking environment.
There is a call to yield() in a loop (see PZEM004T.receive() method), so some background tasks may be running while libaray is waiting for response.

mikekgr, 2018-08-16:
@olehs 
yes, I agree with you that it is hard to follow timing things in multitasking environment.
See some strange in behavior: 
started at 10:44:43 1906ms
11:17:42 2017ms
11:18:42 2020ms
11:19:42 1524ms <== Strange !!!
11:20:42 1528ms
11:21:42 1531ms
11:22:42 1534ms 
...

Please see me sketch above and tell me if it is OK to run on hardware serial, that I did not achieved ...

olehs, 2018-08-16:
It's also possible, that PZEM has it's own timings and requests are processed not as fast as we expect.

> Please see me sketch above and tell me if it is OK to run on hardware serial, that I did not achieved ...

It looks good but I don't have any ESP8266 board available now to check it in hardware. Maybe @vortigont can help as he is the author of HWSerial example...

vortigont, 2018-08-16:
Greetings!
If you want to measure PZEM respone time it would be better to use something like this

```
void ReadAndPost()
{
unsigned long temp;
temp = millis();
float v = pzem.voltage(ip);
float i = pzem.current(ip);
float p = pzem.power(ip);
float e = pzem.energy(ip);
functionDuration = millis() - temp; // moved this line here

if(v >= 0.0){ voltage_blynk = v;} //V
if(i >= 0.0){ current_blynk = i;}  //A
if(p >= 0.0){ power_blynk = p;} //kW
if(e >= 0.0){  energy_blynk = e;} ///kWh

Blynk.virtualWrite(V1, voltage_blynk);
Blynk.virtualWrite(V2, current_blynk);
Blynk.virtualWrite(V3, power_blynk);
Blynk.virtualWrite(V4, (energy_blynk/1000));
}

```

Otherwise all kind of unknown blink activity may interfere.
JFYI: Here is how ESP to PZEM comm works under the oscilloscope
So the whole dialog to retreive 4 meterings takes about 90 ms.

[image]

mikekgr, 2018-08-16:
Dear @vortigont ,
thanks a lot for your kind suggestions, I will do the measurements using your related code. 
By the way when you measure the 4 meterings that for you takes about 90 ms ONLY!, do you use ESP8266 software or hardware serial? Also please see my code above, the section for the hardware serial that I can not get it to work, do you see it right?
I will revert back with the new measurements.
Thanks

vortigont, 2018-08-16:
I use hwserial with ESP8266.
Your code is fine, check your circuits. For ex. if you use wemos with onboard usb2seral converter it may prevent using RX/TX pins with PZEM optocouplers - use swapped pins instead. Consult [WiKi](https://github.com/olehs/PZEM004T/wiki) pages for wiring details and hwserial setup.

mikekgr, 2018-08-17:
Dear @olehs , Dear @vortigont ,
finally, after solving the small hardware glitch that blocked the Hardware Serial operation, ( shame on me, I am 33 years hardware engineer... ), now my setup is running on hardware serial !!! 
Thanks, both of you, for your assistance, and the fine library.
I will stay connected here in case of any problem may arise.

Thanks and Best Regards,
Mike Kranidis

P.S. the hardware problem was the resistor level shiftier... Don't lough please!

closed 2018-08-17 by olehs

vortigont, 2018-08-17:
@mikekgr, that's great! JFYI - SoftwareSerial is indeed has some issues since esp8266 core 2.4.x. There were significant changes in Core code that broke lots of libs based on timings/delay code, new core doesn't like long blocking states in loop(). I think we all better to try writing event driven code without using delays.

vortigont, 2018-08-21:
That's great, @olehs! Event if if not keeping compatibility it would be a great improvement. Feel free to call if beta-testers needed.

mudassar791, 2019-07-20:
hi, 
Respected Member,
how can we connect and take output (Current,Watt,Voltage) from ESP8266 or ESP 32 with Pzem004t, Can any body share coding and wiring schematic?I dont want use OLED,