olehs/PZEM004T #51: Arduino IDE can not compile the sketch.

zimasha, 2018-12-17:
Hi.
After installing esp32 in arduino IDE using the updated method ( not the old git method), the pzem-004t sketches can not be compiled.
It worked before when i was using the old method to install esp32 in arduino IDE.
This is the error:

Arduino: 1.8.3 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default, QIO, 80MHz, 4MB (32Mb), 921600, None"

Archiving built core (caching) in: C:\Users\.......\AppData\Local\Temp\arduino_cache_39693\core\core_esp32_esp32_esp32_PSRAM_disabled,PartitionScheme_default,FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none_65f4f030f519b2015c1c0aff449bc7c2.a
C:\Users\.........\AppData\Local\Temp\arduino_build_60966/arduino.ar(HardwareSerial.cpp.o):(.bss.Serial2+0x0): multiple definition of `Serial2'

sketch\PZEM_ESP32HwSerial.ino.cpp.o:(.bss.Serial2+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ESP32 Dev Module.

olehs, 2018-12-17:
Can you try renaming serial port variable to smth like PzemSerial instead of Serial2 ?

zimasha, 2018-12-17:
Oh. It compiles now... Need to test if it works. Thanks man.