olehs/PZEM004T #52: PZEM004T WITH ARDUINO UNO

saeednama, 2018-12-20:
I have tried the codes here with Arduino UNO, it gives only 0v output doesn't matter if I connect or disconnect the PZEM004T, is this problem from the PZEM or in the code?

olehs, 2018-12-20:
You cannot use pins 0 & 1 on UNO simultaneously with Serial output, because Serial uses these pins.
Try SoftSerial example with different pins.

geev8, 2019-03-27:
Hi am receiving this error msg when compiling my codes make on linux

> 

  /usr/share/arduino//libraries/PZEM004T/PZEM004T.cpp: In member function ‘bool PZEM004T::recieve(uint8_t, uint8_t*)’:
/usr/share/arduino//libraries/PZEM004T/PZEM004T.cpp:148:15: error: ‘yield’ was not declared in this scope
         yield(); // do background netw tasks while blocked for IO (prevents ESP watchdog trigger)
               ^
/usr/share/arduino//libraries/PZEM004T/PZEM004T.cpp:162:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         for(int i=0; i<RESPONSE_DATA_SIZE; i++)
                       ^
/usr/share/arduino/Arduino.mk:1186: recipe for target 'build-uno/libs/PZEM004T/PZEM004T.cpp.o' failed
make: *** [build-uno/libs/PZEM004T/PZEM004T.cpp.o] Error 1

>