olehs/PZEM004T #54: Understanding Power - Energy calculation by PZEM-004T

vandat07, 2019-03-13:
Hello there,

I am using PZEM-004T and Arduino UNO to measure electricity consumption. The module is now working well, I can see the values on Serial screen. However, I still have some questions as below:

**1. How does PZEM-004T calculate the Energy (Wh) from Power (W)?** 

As the following pic: P = 25W, E = 609Wh = 609/3600 Ws. 
With equation: E = P*t, we have t = E/P = 609/3600/25 = 0.006766 s.
So is this calculation correct, if not **please tell me how the module calculates E from P**.

[image]

**2. In the pic, it seems the Energy is calculated instantly. How could we get the accumulated Energy and how to reset it to Zero programmatically?**

Thanks a lot!

Pic sourcce: http://pdacontrolen.com/meter-pzem-004t-with-arduino-esp32-esp8266-python-raspberry-pi/

olehs, 2019-03-13:
1. E is measured in Wh (Watt multiplied by Hour), so 609Wh = 609*3600 Ws.
Here 609Wh means that you were consuming either 609 W during 1 hour, or 25 W during 609/25 = 24.36 hours, or even 1 W during 609 hours and so on.
You equation should be 609*3600/25 = 87696s and it only shows how many seconds is needed to consume 609W with power of 1 W.

2. No, this is an accumulated Energy value and, unfortunately, it cannot be reset programmatically.

vandat07, 2019-03-13:
Thanks a lot for your quick response!

closed 2019-03-18 by olehs

Aliiiu, 2021-02-25:
This isn't still clear enough because what if we want to know energy consumed after 5min. we know the voltage, current reading but not the time

olehs, 2021-02-25:
> This isn't still clear enough because what if we want to know energy consumed after 5min. we know the voltage, current reading but not the time

We also now total energy consumed by now. You just have to subtract Energy value 5 minutes ago from curent Energy value

olehs, 2021-02-25:
Just to make it clear, you have to record values from PZEM with timestamp somewhere (external DB, file etc.)

Aliiiu, 2021-02-25:
Thanks so much for your fast reply, I can't get your first comment. Energy is the accumulation of power over a period of time. Energy after 5 min is power x 5min then you convert to kwh. But we can't control that here. 

concerning your second comment, what is the time interval for each energy calculation made by the meter. I will love to know that can't find it anywhere in the datasheet.

olehs, 2021-02-25:
It's not just about Power multiplied by Time.
Energy is calculated by PZEM's internal MCU using RMS method. It's calculated quite precisely (tens or maybe hundreds of readings per second. I guess...)

Aliiiu, 2021-02-25:
Do you have a link for learning how this energy is calculated. By the way, you will still need time to calculate energy and how do compensate for that.

Aliiiu, 2021-02-25:
[image]
is it something like this

Aliiiu, 2021-02-25:
this is for another metering IC, if you have for pzem-004t pls share

olehs, 2021-02-25:
Sorry, I don't know if PZEM uses special MCU, or just calculates RMS programatically.
You may refer to this article https://learn.openenergymonitor.org/electricity-monitoring/ac-power-theory/arduino-maths
for more information on how to calculate RMS power having only instantaneous voltage and current measurements

olehs, 2021-02-25:
And the Energy will be equal to something like [RMS power] x [number of samples] / [sample rate]

olehs, 2021-02-25:
Found some info on PZEM-004t. Looks like it uses this metering IC
http://www.sdicmicro.com/DataSheet/SD3004%20datasheet%20v0.2c.pdf

Aliiiu, 2021-02-25:
Thanks so much, will look into it

EDSTOBI, 2022-11-11:
hello
I have read through the comments, but I can not implement them.
I have a constant load connected and I get 9.2 W displayed at 0.07A by 234V.
The Wh say 0.76.
somehow all this does not fit together or makes no sense to me

can someone break down my values and explain them to me?