olehs/PZEM004T #58: Does this work with the updated version of PZEM-004T

BrendenCA, 2019-04-27:
Hi,

There is a newer version of the PZEM-004T
It uses the Modbus–RTU, will this library work for it?

https://www.aliexpress.com/store/product/PZEM-004T-Update-Version-AC-Single-Phase-Multifunction-Ampere-Meter-Watt-Power-Volt-Amp-Kwh-TTL/1773456_32907139707.html?spm=2114.12010612.8148356.2.a5605ff1fOKeWd

Thanks,

Giardo84, 2019-05-08:
> Hi,
>
> There is a newer version of the PZEM-004T
> It uses the Modbus–RTU, will this library work for it? [...]

I've tried but it doesn't work

schreda, 2019-05-09:
Is there a working library out there already?

SeByDocKy, 2019-05-10:
Desperatly need a V3 support :( :( :( :( .....

xcusso, 2019-06-01:
The new n version is working with pzem-016, but I don't know why..

SumeriLal, 2019-08-28:
Use this library: https://github.com/SumeriLal/Arduino/tree/master/MasterPZEM

mandulaj, 2019-12-17:
Hi, a bit late, but here is another one too :smile: 
https://github.com/mandulaj/PZEM-004T-v30

closed 2020-07-25 by olehs