olehs/PZEM004T #60: PZEM004T V3.0 not compatible with this library

aksh2412, 2019-06-28:
Hello,

I bought new PZEM004T V3.0 module and using this library it is was not working, on further debugging found that the command sequence for PZEM004T V3.0 and PZEM004T V1.0 are different.
Please let me know if my above observation is correct.

Also provide me with the solution as PZEM004T V3.0 requires CRC of 16 bit with 0xA001 polynomial.

Thank you,
Akshay

RuslanKap, 2019-06-28:
+

aksh2412, 2019-06-28:
Can you suggest any solution for this?

innovators-guru, 2019-06-30:
Check all details about PZEM-004T V3.0 and its library  https://innovatorsguru.com/pzem-004t-v3/

innovators-guru, 2019-06-30:
Check all details about PZEM-004T V3.0 and its library  [https://innovatorsguru.com/pzem-004t-v3/]

SumeriLal, 2019-08-28:
Use this library : https://github.com/SumeriLal/Arduino/tree/master/MasterPZEM

mandulaj, 2019-12-17:
I am in a bit of conflict of interest, but my library isn't half bad https://github.com/mandulaj/PZEM-004T-v30  :smile:

mandulaj, 2019-12-17:
Also this is a duplicate of #58

closed 2020-07-25 by olehs