olehs/PZEM004T #61: Serial monitor cannot read the value from PZEM through nodemcu esp8266

msyth333, 2019-09-21:
hi.. I had been connecting pzem to nodemcu. but the issue was my serial monitor can't read the reading even though I successfully upload my file. They just display the unit but when I turn on main supply and load, they did not detect the value. Here was my hardware connection, result and coding. Hopefully someone can help me to figure out this issue. 
[image]
[image]
[image]
[image]

morganflint, 2020-07-26:
You are using PZEM004T V3.0 (the same I have, with white optocouplers). The communication protocol is different from the one in the previous version, so this library is not compatible with V3.

This is the manual for V3, including the protocol description:
https://innovatorsguru.com/wp-content/uploads/2019/06/PZEM-004T-V3.0-Datasheet-User-Manual.pdf

The manual for the older version is this one, which describes the same protocol as the readme in this library:
https://www.circuitspecialists.com/content/189799/ac004.pdf
(note that although it says "V3.0" It refers to PZEM-004, not PZEM-004T)

morganflint, 2020-07-26:
In this other issue:
https://github.com/olehs/PZEM004T/issues/40
SumeriLal pointed to another library, specific for 004T V3.0:
https://github.com/SumeriLal/Arduino/tree/master/MasterPZEM

morganflint, 2020-07-26:
Another library for PZEM004T-V3.0:
https://github.com/mandulaj/PZEM-004T-v30

closed 2021-06-02 by olehs

margohaux, 2022-11-18:
### I had been connecting pzem to TTGO ESP32. but the issue was my serial monitor can't read the reading even though I successfully upload my file. They just display the unit but when I turn on main supply and load, they did not detect the value.. Hopefully someone can help me to figure out this issue.

#include <PZEM004Tv30.h>
#include <SoftwareSerial.h>
#include <Wire.h>
#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd(0x27, 16, 2);

#if defined(ESP32)
PZEM004Tv30 pzem(Serial1, TX, RX);
#else
PZEM004Tv30 pzem(Serial1);
#endif

void setup() {
/* Debugging serial */
Serial.begin(9600);
lcd.init();
lcd.backlight();
lcd.setCursor(0, 0);
lcd.setCursor(0, 1);
lcd.print("ENERGY METER");
delay(500);
lcd.clear();

}

void loop() {

float voltage = pzem.voltage();
  if(voltage != NAN){
    Serial.print("Voltage: ");
    Serial.print(voltage);
    Serial.println("V");
    lcd.setCursor(0,0);
    lcd.print("V:");
    lcd.print("VOLTAGE");
  
  }else{
    Serial.println("Error reading voltage");
  }
            

  float current = pzem.current();
  if (current != NAN){
    Serial.print("Current: ");
    Serial.print(current);
    Serial.println("A");
    lcd.setCursor(0,1);
    lcd.print("I:");
    lcd.print("CURRENT");
    
  }else{
    Serial.println("Error reading current");
    }

    
  float power = pzem.power();
  if (power != NAN) {
    Serial.print("Power: ");
    Serial.print(power);
    Serial.println("W");
    lcd.setCursor(8,0);
    lcd.print("P:");
    lcd.print("POWER");
  
  }else{
    Serial.println("Error reading power");
  }
  
  
  float energy = pzem.energy();
  if (energy != NAN) {
    Serial.print("Energy: ");
    Serial.print(energy,3);
    Serial.println("kWh");
  }else{
    Serial.println("Error reading energy");
  }
  
  float frequency = pzem.frequency();
  if (frequency != NAN) {
    Serial.print("Frequency: ");
    Serial.print(frequency, 1);
    Serial.println("Hz");
    lcd.setCursor(8,1);
    lcd.print("F:");
    lcd.print("FREQUENCY");
         
  }else{
    Serial.println("Error reading frequency");
  }
  

  float pf = pzem.pf();
  if (pf != NAN) {
    Serial.print("PF: ");Serial.println(pf);
    lcd.setCursor(0,3);
    lcd.print("PF:");
    lcd.print("pf");
  }else{
    Serial.println("Error reading power factor");
  } 
Serial.println();
delay(2000);
}

SERIAL MONITOR: Energy:

Voltage: nanV
Current: nanA
Power: nanW
Energy: nankWh
Frequency: nanHz
PF: nan

olehs, 2022-11-18:
@margohaux Wrong repo
https://github.com/mandulaj/PZEM-004T-v30

margohaux, 2022-11-18:
i tried the code for the hardware serial for my ttgo tcall esp32 but i get an error, error reading.