olehs/PZEM004T #63: Bidirectional Current

thiagoarreguy, 2020-05-31:
Its possible to read a current and know the way flows?

For example, to monitoring a solar photovoltaic system that send current to the grid.

I have found the device on aliexpress and i wish to one with same functions: 

https://www.aliexpress.com/item/10000025557485.html

I know that 6 in 1 meter have an arrow on CT to measure the current on correct way:
https://a.aliexpress.com/_dWFRgEX

Bu i dont know if this CT will work with pzem-004t.

Thanks in advanced.

thiagoarreguy, 2020-06-01:
pzem-004t allow negative power value? if the current is negative...

vortigont, 2020-06-01:
@thiagoarreguy, it is not possible out of the box, Although I remember it was mentioned that PZEM controller chip has a dedicated pin that represents the direction of the current flow, but it is nither traced nor processed in software. The coil itself doesn't matter.
I believe it would be much simpler to install a dedicated coil and meter to the solar inverter itself and measure the output power there.

thiagoarreguy, 2020-06-01:
I already did: 

http://thiagoarreguy.servehttp.com/arquivos/proxy.php?page=energia

http://thiagoarreguy.servehttp.com/arquivos/graficopotencia.php?date=today

But i would like to know the amount of energy injected on the grid (during daylight) and energy consumption  (during night).

vortigont, 2020-06-01:
PZEM has only one omnidirectional cumulative counter, it doesn't segregate IN and OUT energy into two counters. 1kWh IN and 1kWh OUT would result in 2 kWh absolute energy.
Even if you do HW mod and read pin status you would have to do the SUM math your self. Better to look for other powermeters out there with required features.

thiagoarreguy, 2020-06-01:
This 6 in 1 energy meter measure the current only in one way:
https://a.aliexpress.com/_dWFRgEX

I wish have a pzem-004t with the same function =)

vortigont, 2020-06-02:
Why do you think so? I have one of these, it is the same as PZEM - doesn't matter with which wire you feed it Live or Neutral it always increments.

thiagoarreguy, 2020-06-02:
Not this model, is not from peacefear.
And have an arrow on CT to measure the current on correct way.

closed 2021-06-02 by olehs