olehs/PZEM004T #64: Multiple Pzems

efakostya, 2020-07-25:
Hello friends, tell me how to fix the library to read in a loop.
I don't like the case option.
PZEM004T pzem1 (D1, D2);
PZEM004T pzem2 (D3, D4);
PZEM004T pzem3 (D5, D6);

v [0] = pzem1.voltage ();
c [0] = pzem1.current ();
p [0] = pzem1.power ();
e [0] = pzem1.energy ();
f [0] = pzem1.frequency ();

v [1] = pzem2.voltage ();
c [1] = pzem2.current ();
p [1] = pzem2.power ();
e [1] = pzem2.energy ();
f [1] = pzem2.frequency ();

v [2] = pzem3.voltage ();
c [2] = pzem3.current ();
p [2] = pzem3.power ();
e [2] = pzem3.energy ();
f [2] = pzem3.frequency ();

Is it possible to somehow bring to the form:
for (size_t i = 0; i < 3; i++){
v [i] = pzem[i].voltage ();
c [i] = pzem[i].current ();
p [i] = pzem[i].power ();
e [i] = pzem[i].energy ();
f [i] = pzem[i].frequency ();
}

olehs, 2020-07-25:
You can use an array of pointers
```C
PZEM004T* pzem[] = {
  new PZEM004T(D1, D2),
  new PZEM004T(D3, D4),
  new PZEM004T(D5, D6)
}
...
for (size_t i = 0; i < 3; i++){
  v[i] = pzem[i]->voltage();
  ...
}
```

efakostya, 2020-07-26:
Thank you so much for the hint.  Your library helped me a lot in creating the application.

Огромное спасибо Вам за подсказку. Ваша библиотека очень помогла мне в создании приложения.

closed 2020-07-26 by efakostya

crizangel, 2020-08-02:
> Thank you so much for the hint. Your library helped me a lot in creating the application.

Hello, can I take a look at your code? I can't seem to make mine work (2 of v2.0)