olehs/PZEM004T #65: Cannot get Multiple PZEM004Ts to work

crizangel, 2020-08-02:
Hello, I was wondering if anyone else has some code that made multiple PZEM004Ts to work? I can't seem to make 2 work at the moment, I plan to incorporate 9 PZEM004Ts into my project, and I am starting out with 2 for the meantime, in order to make a proof of concept of my project.

The devices that I am using are 9 PZEM004T (1 - V1.0, 6 - V2.0, and 2 - V3.0), Arduino Mega 2560, Fotek SSR-60DA Solid State Relays, and some electrical outlets.

I am trying my very best to make only two work but all the articles that I have come across are not working for me. I have tried using different libraries, mixing and matching power analyzers, and using example codes from other articles and forum posts.

Might I also suggest adding an example code in the library that utilizes 2 power analyzers and a guide on how to make multiple ones work? Many thanks!

Attached is a slightly modified code that I acquired from https://github.com/olehs/PZEM004T/issues/10 (PZEM_Local2Instances.txt)

`#include <SoftwareSerial.h>
#include <PZEM004T.h>

#define txMain 11 //Single pin for all Meters
#define rx1 10    //Meter 1
#define rx2 9     //Meter 2

//PZEM004T pzem1(rx1, txMain); // RX,TX
IPAddress ip1(192, 168, 1, 1);

//PZEM004T pzem2(rx2, txMain); // RX,TX
IPAddress ip2(192, 168, 1, 10);

void setup() {
  Serial.begin(9600);
  // pzem.setAddress(ip);
}

void loop() {
  
  PZEM004T pzem1(rx1, txMain); // RX,TX
  
  Serial.println ("METER 1:");
  
  float v1 = pzem1.voltage(ip1);
  if (v1 < 0.0) v1 = 0.0;
  Serial.print(v1); Serial.print("V1; ");
  
  float i1 = pzem1.current(ip1);
  if (i1 >= 0.0) {
    Serial.print(i1);
    Serial.print("A1; ");
  }
  
  float p1 = pzem1.power(ip1);
  if (p1 >= 0.0) {
    Serial.print(p1);
    Serial.print("W1; ");
  }

  float e1 = pzem1.energy(ip1);
  if (e1 >= 0.0) {
    Serial.print(e1);
    Serial.print("Wh1; ");
  }
  
  Serial.println();

delay(1500);
  
  PZEM004T pzem2(rx2, txMain); // RX,TX
  
  Serial.println ("METER 2:");
  
  float v2 = pzem2.voltage(ip2);
  if (v2 < 0.0) v2 = 0.0;
  Serial.print(v2); Serial.print("V2; ");
  
  float i2 = pzem2.current(ip2);
  if (i2 >= 0.0) {
    Serial.print(i2);
    Serial.print("A2; ");
  }
  
  float p2 = pzem2.power(ip2);
  if (p2 >= 0.0) {
    Serial.print(p2);
    Serial.print("W2; ");
  }

  float e2 = pzem2.energy(ip2);
  if (e2 >= 0.0) {
    Serial.print(e2);
    Serial.print("Wh2; ");
  }
  
  Serial.println();

  delay(1500);
}`

shiftux, 2021-09-13:
@crizangel i'd be interested in knowing whether you have been able to make this work...
thanks and best

vortigont, 2021-09-14:
@crizangel I was able to run multiple PZEM v3.0's over single UART port with a bit of hacking for current libs. Though I decided to write a new lib from scratch specially designed to run multiple devices. Currently still in progress. I'm going to make it publicly available once ready.

vortigont, 2021-09-24:
@shiftux invite you test a new pzem lib [project](https://github.com/vortigont/pzem-edl) designed to run multiple pzem instances.
Any feedback would be appreciated.