olehs/PZEM004T #67: PZEM-004 vs PZEM-004T

ipa64, 2020-12-08:
Hi,
I have these two versions with their documentations.
At first I thought they are the same, 004 has display 7seg, 004T doesn’t have display.
But it’s wrong, the protocol between 004 and 004T is different, 004T I use Modbus-RTU
And the protocol you describe in your readme match to the PZEM-004.
So your library should be named PZEM-004
Or maybe they have changed the protocol without modify the name, I hope not ???
There is on internet a lot of confusion with this two modules.

olehs, 2020-12-08:
Hi, I guess they change versions as they want (
The library was written for the board like on the image below. It says "V1.0" - maybe that's the key.
[image]

ipa64, 2020-12-09:
I just got information from Peacefair 
From 004T v3.0 protocol is Modbus-RTU, previous version was not !
So they changed the protocol without modifying the name ! So bad!

ipa64, 2020-12-09:
And the icing on the cake:
PZEM-004 v3.0 with display doesn’t  have Modbus-RTU protocol !

closed 2021-06-02 by olehs