olehs/PZEM004T #70: Compiler error when used with Arduino UNO Wifi REV 2

MikeHales1, 2021-03-02:
When attempting to use with the Arduino UNO Wifi Rev 2 module, the compiler generates the following errors:

O:\Documents\Arduino\libraries\PZEM004Tv30\PZEM004Tv30.cpp:54:23: error: conflicting declaration 'arduino::HardwareSerial Serial'
 extern HardwareSerial Serial;
                       ^~~~~~
In file included from C:\Users\Mike\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\cores\arduino/Arduino.h:140:0,
                 from O:\Documents\Arduino\libraries\PZEM004Tv30\PZEM004Tv30.h:47,
                 from O:\Documents\Arduino\libraries\PZEM004Tv30\PZEM004Tv30.cpp:23:
C:\Users\Mike\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\cores\arduino/UART.h:182:20: note: previous declaration as 'UartClass Serial'
   extern UartClass Serial;
                    ^~~~~~
exit status 1
Error compiling for board Arduino Uno WiFi Rev2.

I've previously compiler to Arduino MEGA without any problems. I used the Example file 'PZEMHardSerial' with Serial3 changed to Serial1.

Please advise. Thanks.

olehs, 2021-03-02:
PZEM004T**v30** ?
Are you sure this is the right repo ? ;)

MikeHales1, 2021-03-03:
Olehs, Sorry you are correct.

closed 2021-03-03 by MikeHales1