olehs/PZEM004T #75: Reallocate hardware serial port in ESP32 Heltec LoRa V2

ManuelJimenezBuendia, 2021-06-02:
Would it be possible to add an additional constructor to map a hw serial port to new pins? It wouldpar be ticularly useful when using Heltec ESP32 LoRa V2 to map Serial2 to two of the few available pins on the board (e.g. 37,13).

In PZEM-004Tv30.h, additional public constructor:
PZEM004Tv30(HardwareSerial* port, int rx_pin, int tx_pin, uint8_t addr=PZEM_DEFAULT_ADDR);

In PZEM-004Tv30.cpp: 
PZEM004Tv30::PZEM004Tv30(HardwareSerial* port, int rx_pin, int tx_pin, uint8_t addr)
{
    port->begin(PZEM_BAUD_RATE, SERIAL_8N1, rx_pin, tx_pin);  //Baud rate, parity mode, RX, TX
    this->_serial = port;
    this->_isSoft = false;
    init(addr);
}

So in the sketch, the instance of the class would be created with two additional parameters for the pins:
PZEM004Tv30 pzem(&Serial2, 37, 13); // (serial port, RX pin, TX pin)

I have checked the library with these modifications and it works fine.

Thanks!

olehs, 2021-06-02:
There is constructor for pin mapping on ESP32.
Are you sure you're using exactly this lib? Looks like you're using library for v3.0 of PZEM

closed 2021-06-04 by olehs

ManuelJimenezBuendia, 2021-06-04:
You're right, I was using the mandulaj/PZEM-004T-v30. Sorry for the mistake.

vortigont, 2021-06-04:
@ManuelJimenezBuendia there is a [PR](https://github.com/mandulaj/PZEM-004T-v30/pull/55) pending for this feature. You can use mine [branch](https://github.com/vortigont/PZEM-004T-v30/tree/vortigont) meanwhile if you like

ManuelJimenezBuendia, 2021-06-04:
You're right, I will use yours.
Thanks!!!